slic3r/Slic3r #4750: Elephant foot compensation can lead to unprintable models

probonopd, 2019-03-02:
With Elephant foot compensation enabled, this

[image]

becomes this

[image]

which results in an unprintable object because the second layer will be printed in mid air.

The solution could be to restrict elephant foot compensation to the __outermost perimeter__ of the model.

Reference: https://github.com/prusa3d/Slic3r/issues/1757

VanessaE, 2019-03-02:
I would suggest just limiting it such that walls aren't reduced below the first-layer line width.

supermerill, 2019-03-02:
Also, you can active thin walls.

probonopd, 2019-03-02:
"Detect thin walls" is on by default.

lordofhyphens, 2019-03-09:
Which setting is "elephant foot compensation" supposed to be?

supermerill, 2019-03-10:
it's a slic3rPE feature, like XY compensation, but only for the first layer.

lordofhyphens, 2019-03-10:
So why is this here in the main Slic3r issue tracker?

probonopd, 2019-03-10:
Oops, I didn't realize that Slic3r doesn't have this feature. (The whole situation with 2 almost-same applications is a bit confusing.) Sorry!

In this case, I change the story to be a feature request for "elephant foot compensation".

lordofhyphens, 2019-03-11:
So you are looking for the following user story:
As a 3d printer user, I want to be able to shrink the size of the first layer some fixed mm so that I can compensate for issues with my printer's first layer being wider than I want.

As a user of this feature, I want Slic3r to limit the amount of compensation so that thin walls on the first layer are not removed, so that I do not have a situation where layers 2+ print in the air.

The second one will increase the complexity of the feature by quite a bit, as the situation can be hard to detect in the general case and we need to define what we think the correct behavior is in this case.

lordofhyphens, 2019-03-11:
@probonopd I have a good idea how this was probably implemented in the Prusa3d fork and it would be a hack at best (basically a special case of the X/y compensation only applied to whatever the first layer height is set to).

lordofhyphens, 2019-03-11:
https://github.com/slic3r/Slic3r/blob/70946af2ae0fa0be69b16defbe14ce72d71071e4/xs/src/libslic3r/PrintObject.cpp

The basic hack would be easy to apply there in PrintObject.cpp, basically create a new advanced option and apply it only to layer id 0. Test should not be too bad either.

How to handle the undesired case is another question that needs more discussion.