slic3r/Slic3r #4751: Slic3r segfaults on stl from thingiverse

kefir-, 2019-03-02:
**Describe the bug**
Slic3r segfaults when I try to add the STL from https://www.thingiverse.com/download:3266461 (Shuriken.stl)

**To Reproduce**
Steps to reproduce the behavior:
1. Download https://www.thingiverse.com/download:3266461
2. Open Slic3r
3. Try adding the downloaded STL (Shuriken.stl)
4. Observe segfault

**3D Model and Slic3r Configuration Export**
Config: https://www.dropbox.com/s/nbpr17mr8a38yx6/slic3r-config-PLA-coldbed.ini?dl=0
STL: https://www.thingiverse.com/download:3266461

**Expected behavior**
No segfault, please. 

**Desktop:**
Tested on recent git revision 05a2f9028a35c56e039cf7a531d63b7ff7e97645 on Ubuntu Linux 18.04.

kefir-, 2019-03-02:
Just to add, I can easily fix the STL by opening it in Meshlab, and then saving a new STL from Meshlab. I clicked "yes" to "Unify duplicated vertices" in meshlab, but besides that no action is necessary other than exporting to a new STL.

VanessaE, 2019-03-02:
I've had many a crash trying to load STLs, or if a model loads, trying to "cut" it may also cause a crash.

If a model is "garbage", that's fine..  it should just look like it when sliced :stuck_out_tongue_closed_eyes:

lordofhyphens, 2019-03-02:
If model is garbage, probably not a bug in Slic3r, but the crash is probably incurred in admesh.

VanessaE, 2019-03-02:
Well, it's kind of a personal standard -- a hard crash is *always* a bug, regardless of how it came about.  In this instance, it's better to throw a warning (if possible) and have a junk slicing result, than a crash. :smiley:

kefir-, 2019-03-02:
Is the model garbage? All I did in meshlab was to unify vertices, as mentioned, and after that Slic3r gives no indication that anything is wrong. I tried again now, and avoided the unify vertices step, exported the mesh, and it still looks fine in Slic3r.

Looking at the original STL file, it appears to have been generated by Cura. I say that because it starts with the string "CURA BINARY STL EXPORT. Sat 21 Jan 2017 00:54:00".

I didn't know slic3r used admesh, is it the one from https://github.com/admesh/admesh? I fetched the latest master branch of admesh from https://github.com/admesh/admesh.git and built that, and then I ran the following command. The resulting file, test4.stl, works fine in Slic3r. test.stl is the same file as Shuriken.stl, as you can see from the header printed.

```
$ admesh -b test4.stl test.stl 
ADMesh version 0.99.0dev, Copyright (C) 1995, 1996 Anthony D. Martin
ADMesh comes with NO WARRANTY.  This is free software, and you are welcome to
redistribute it under certain conditions.  See the file COPYING for details.
Opening test.stl
Checking exact...
Checking nearby. Tolerance= 0.084300 Iteration=1 of 2...  Fixed 0 edges.
Checking nearby. Tolerance= 0.087473 Iteration=2 of 2...  Fixed 0 edges.
Removing unconnected facets...
Filling holes...
Checking normal directions...
Checking normal values...
Calculating volume...
Calculate surfacea area...
Verifying neighbors...
Writing binary file test4.stl

================= Results produced by ADMesh version 0.99.0dev ================
Input file         : test.stl
File type          : Binary STL file
Header             : CURA BINARY STL EXPORT. Sat 21 Jan 2017 00:54:00
============== Size ==============
Min X = -15.639952, Max X =  6.399748
Min Y = -22.200089, Max Y = -0.160288
Min Z =  0.000000, Max Z =  5.936300
========= Facet Status ========== Original ============ Final ====
Number of facets                 :  2882                2974
Facets with 1 disconnected edge  :    98                   0
Facets with 2 disconnected edges :     0                   0
Facets with 3 disconnected edges :     0                   0
Total disconnected facets        :    98                   0
=== Processing Statistics ===       ====== Other Statistics ======
Number of parts       :     3       Volume          :    919.067688
Degenerate facets     :     4       Surface area    :    1556.031616
Edges fixed           :     0
Facets removed        :     4
Facets added          :    96
Facets reversed       :    96
Backwards edges       :     0
Normals fixed         :  2974
```

lordofhyphens, 2019-03-24:
@kefir- yes, we load the vertices and faces from admesh into a TriangleMesh structure in libslic3r.

lordofhyphens, 2019-05-18:
@kefir- it's very close to the upstream; I think we have a few patches on it that aren't upstream, mostly to avoid compiler warnings, etc.

It's crashing during the stl_fill_holes routine, even with my PR #4805 to just do what admesh itself would use. I haven't tracked it down further than that.

@hroncok Any thoughts?

hroncok, 2019-05-18:
Quite a huge diff actually:

```patch
diff -u xs/src/admesh/connect.c admesh-org/admesh/src/connect.c
--- xs/src/admesh/connect.c	2019-05-18 21:13:51.077662806 +0200
+++ admesh-org/admesh/src/connect.c	2017-03-20 09:32:05.688102361 +0100
@@ -82,16 +82,6 @@
 
   for(i = 0; i < stl->stats.number_of_facets; i++) {
     facet = stl->facet_start[i];
-    // Positive and negative zeros are possible in the floats, which are considered equal by the FP unit.
-    // When using a memcmp on raw floats, those numbers report to be different.
-    // Unify all +0 and -0 to +0 to make the floats equal under memcmp.
-    {
-      uint32_t *f = (uint32_t*)&facet;
-      for (int j = 0; j < 12; ++ j, ++ f) // 3x vertex + normal: 4x3 = 12 floats
-        if (*f == 0x80000000)
-            // Negative zero, switch to positive zero.
-            *f = 0;
-    }
 
     /* If any two of the three vertices are found to be exactally the same, call them degenerate and remove the facet. */
     if(   !memcmp(&facet.vertex[0], &facet.vertex[1],
@@ -116,42 +106,53 @@
     }
   }
   stl_free_edges(stl);
-
-#if 0
-  printf("Number of faces: %d, number of manifold edges: %d, number of connected edges: %d, number of unconnected edges: %d\r\n", 
-    stl->stats.number_of_facets, stl->stats.number_of_facets * 3, 
-    stl->stats.connected_edges, stl->stats.number_of_facets * 3 - stl->stats.connected_edges);
-#endif
 }
 
 static void
 stl_load_edge_exact(stl_file *stl, stl_hash_edge *edge,
                     stl_vertex *a, stl_vertex *b) {
 
+  float diff_x;
+  float diff_y;
+  float diff_z;
+  float max_diff;
+
   if (stl->error) return;
 
-  {
-    float diff_x = ABS(a->x - b->x);
-    float diff_y = ABS(a->y - b->y);
-    float diff_z = ABS(a->z - b->z);
-    float max_diff = STL_MAX(diff_x, diff_y);
-    max_diff = STL_MAX(diff_z, max_diff);
-    stl->stats.shortest_edge = STL_MIN(max_diff, stl->stats.shortest_edge);
-  }
-
-  // Ensure identical vertex ordering of equal edges.
-  // This method is numerically robust.
-  if ((a->x != b->x) ? 
-        (a->x < b->x) : 
-        ((a->y != b->y) ? 
-            (a->y < b->y) : 
-            (a->z < b->z))) {
-    memcpy(&edge->key[0], a, sizeof(stl_vertex));
-    memcpy(&edge->key[3], b, sizeof(stl_vertex));
-  } else {
-    memcpy(&edge->key[0], b, sizeof(stl_vertex));
-    memcpy(&edge->key[3], a, sizeof(stl_vertex));
-    edge->which_edge += 3; /* this edge is loaded backwards */
+  diff_x = ABS(a->x - b->x);
+  diff_y = ABS(a->y - b->y);
[... 1042 lines cut ...]
-  if(stl->stats.volume < 0.0) {
-    stl_reverse_all_facets(stl);
-    stl->stats.volume = -stl->stats.volume;
-  }
 }
 
 static float get_area(stl_facet *facet) {
@@ -565,6 +540,15 @@
     printf("Calculating volume...\n");
   stl_calculate_volume(stl);
 
+  if(fixall_flag) {
+    if(stl->stats.volume < 0.0) {
+      if (verbose_flag)
+        printf("Reversing all facets because volume is negative...\n");
+      stl_reverse_all_facets(stl);
+      stl->stats.volume = -stl->stats.volume;
+    }
+  }
+
   if(exact_flag) {
     if (verbose_flag)
       printf("Verifying neighbors...\n");

``

lordofhyphens, 2019-05-18:
Half of it is just using wchar_t on Windows (all of the prototype changes).

Honestly I mostly want to point to admesh as a submodule and be done with it, but the perl build system wants to try to compile everything in the hierarchy (because it is not terribly bright).

lordofhyphens, 2019-05-18:
I'm looking through the diffs now and piecing together what looks like changes from our side via `git blame`.

lordofhyphens, 2019-05-18:
@hroncok the fp zero unification looks like it got pulled from admesh at some point, any idea why?

hroncok, 2019-05-18:
I don't recall, that would need to be blamed as well.

I'm all for merging stuff into one project. I haven't got time for ADMesh for a very long time unfortunately, but I can review PRs.

lordofhyphens, 2019-05-18:
I've copied over admesh https://github.com/slic3r/Slic3r/tree/update-admesh, leaving just a few minor changes in the header.

Any idea what I am doing wrong here?

````
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c46535 in __GI_abort () at abort.c:79
#2  0x00007ffff7c9d508 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7da828d "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff7ca3c1a in malloc_printerr (str=str@entry=0x7ffff7da656a "realloc(): invalid next size")
    at malloc.c:5341
#4  0x00007ffff7ca7da4 in _int_realloc (av=av@entry=0x7ffff7ddfc40 <main_arena>, 
    oldp=oldp@entry=0x555556720de0, oldsize=oldsize@entry=149664, nb=nb@entry=163184) at malloc.c:4561
#5  0x00007ffff7ca8d4f in __GI___libc_realloc (oldmem=0x555556720df0, bytes=163176) at malloc.c:3213
#6  0x00007ffff72f931a in stl_add_facet (new_facet=0x7fffffffe240, stl=0x5555564ec7d0)
    at src/admesh/connect.c:956
#7  stl_add_facet (stl=0x5555564ec7d0, new_facet=0x7fffffffe240) at src/admesh/connect.c:951
#8  0x00007ffff72f972c in stl_fill_holes (stl=stl@entry=0x5555564ec7d0) at src/admesh/connect.c:922
#9  0x00007ffff7272199 in stl_repair (stl=0x5555564ec7d0, fixall_flag=<optimized out>, 
    exact_flag=<optimized out>, tolerance_flag=<optimized out>, tolerance=<optimized out>, 
    increment_flag=<optimized out>, increment=0.00317291799, nearby_flag=<optimized out>, iterations=10, 
    remove_unconnected_flag=1, fill_holes_flag=1, normal_directions_flag=1, normal_values_flag=1, 
    reverse_all_flag=0, verbose_flag=0) at src/admesh/util.c:537
#10 0x00007ffff748999e in Slic3r::TriangleMesh::repair (this=0x5555564ec7d0)
    at src/libslic3r/TriangleMesh.cpp:180
#11 Slic3r::TriangleMesh::repair (this=0x5555564ec7d0) at src/libslic3r/TriangleMesh.cpp:168
#12 0x00007ffff74b8a78 in Slic3r::ModelObject::repair (this=0x5555562bc530) at src/libslic3r/Model.cpp:604
#13 0x00007ffff74b8ab4 in Slic3r::Model::repair (this=0x5555565698f0) at src/libslic3r/Model.cpp:213
#14 0x00007ffff71503df in XS_Slic3r__Model_repair (my_perl=<optimized out>, cv=<optimized out>)
    at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/home/lenox/slic3r/xs/xsp/typemap.xspt"  "/home/lenox/slic3r/xs/xsp/Model.xsp":465
````

hroncok, 2019-05-19:
I don't know. Was hit once by a similar problem in numpy-stl (Python), it was locale specific? Can you reproduce with LC_ALL=C?

poikilos, 2021-01-22:
Here is another mesh with `realloc(): invalid next size`
Full output on 1.3.1-dev nightly from 2021-01-22 updated in the morning in Eastern Time:
```
/home/owner/.local/lib64/Slic3r-master-latest.AppImage
Gtk-Message: 10:22:03.978: Failed to load module "xapp-gtk3-module"

** (slic3r.pl:189524): WARNING **: 10:22:04.539: invalid source position for vertical gradient

** (slic3r.pl:189524): WARNING **: 10:22:04.539: invalid source position for vertical gradient

** (slic3r.pl:189524): WARNING **: 10:22:10.824: invalid source position for vertical gradient

** (slic3r.pl:189524): WARNING **: 10:22:10.824: invalid source position for vertical gradient
realloc(): invalid next size
/tmp/.mount_Slic3rvA2PbL/AppRun: line 65: 189524 Aborted                 (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib:${LD_LIBRARY_PATH}" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"
```

[SHfCCaF - Dell GT730 duct for 90mm fan.zip](https://github.com/slic3r/Slic3r/files/5856812/SHfCCaF.-.Dell.GT730.duct.for.90mm.fan.zip)
- See also #3317
- See also #4053
- See also #4253