slic3r/Slic3r #4752: Set infill overlap separately for first layer?

ToxicFrog, 2019-03-03:
**Is your feature request related to a problem? Please describe.**
Interior layers seem to print fine, but I'm having issues with the last and, especially, the first layer infill not properly contacting the walls, as seen in the first two of [these images](https://ancilla.ancilla.ca/share/e56zXQVhAn08kBq5FRzzINRRjRSPQOxc8R5DvnS7TD8=/). I'm confident that this isn't a problem with the printer hardware because it [doesn't happen when I use Cura](https://ancilla.ancilla.ca/share/e56zXQVhAn08kBq5FRzzINRRjRSPQOxc8R5DvnS7TD8=/#2), but in general I prefer Slic3r and would like to keep using it. The photos are using Archimedean Chords but it happens with every infill pattern.

**Describe the solution you'd like**
The first layer doesn't seem to respect the "infill/perimeters overlap" setting, and even if it did I want relatively little overlap for the interior (15-20% seems to be the sweet spot for this printer, the default of 55% produces visible bulges in the walls of the print). But it seems like cranking it up to, say, 50-60% for the first layer might improve matters a bit.

**Describe alternatives you've considered**
I've tried playing around with first layer thickness, extrusion width, speed, infill patterns, extruder temperature, and the infill/perimeters overlap with no success; some of them make visible differences in e.g. the line width and density of the slice but without actually connecting the infill to the perimeter reliably.

supermerill, 2019-03-03:
Imo, the three pictures show heavily under-extruded objects. Did/can you try to print the same gcode (cura & slic3r) with 120% flow?

ToxicFrog, 2019-03-03:
As in, set `Filament -> Filament -> Extrusion Multiplier` to 1.2 and try again? I can do that, sure.

The Cura I'm using is the Cura 15 Octoprint plugin, not Cura-on-the-desktop, so it doesn't expose that tuning knob, but I looked at the printer profile and the Cura one was also printed at 100% flow rate.

If that solves it, is the long term fix to adjust the E steps-per-mm calibration on my printer?

ToxicFrog, 2019-03-03:
[Bottom view](https://ancilla.ancilla.ca/share/e56zXQVhAn08kBq5FRzzINRRjRSPQOxc8R5DvnS7TD8=/#3). Top one was printed with extrusion multiplier=1.0, bottom one with 1.2; identical settings otherwise. The bottom one looks slightly better, but only slightly. Can my E-calibration really be off by that much? XYZ needed to be adjusted by about 1%.

[Top view](https://ancilla.ancilla.ca/share/e56zXQVhAn08kBq5FRzzINRRjRSPQOxc8R5DvnS7TD8=/#4). The one printed with 1.0 has a visible gap a few layers from the top. The one printed with 1.2 trades that for a weird globby "crown" of filament, although that might just be drippings as the extruder retracted.

ToxicFrog, 2019-03-03:
Did some tests with a pair of calipers and the octoprint terminal. The filament is configured as 1.75mm and measures as around 1.78mm. `G1 E-10` wound it back 9mm. So it may be underextruding by about 10% based on that, but in that case I'd expect setting the flow rate to 1.2 to fix it and then some.

supermerill, 2019-03-03:
You may want to lower your z, maybe it's printing the first layer too high.
I recommend to print some first-layer calibration pattern (some 1-layer flat rectangle in each corner & the center). See [here for an example](https://help.prusa3d.com/article/ZhBlGFD9Ah-live-adjust-z)