slic3r/Slic3r #4753: SLic3r ouput has part that is not part of model

athulsnair, 2019-03-04:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
I just created a model with a wall thickness of 0.5 mm. Sliced in slic3r with extrusion width set as 0.5(Tried AUTO as well). But when finished slicing, output has a portion that is not part of original model. I don;t know is it a bug or some default set up.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice the attached model
2. Compare STL with Slic3r ouput

**Expected behavior**
It should be same as that of the model. At first I thought it would be some support. But it's not.

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 1.3.1-dev
     - _Do not report Prusa3D Slic3r bugs here without confirmin g it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

lordofhyphens, 2019-03-04:
Repair the model.

athulsnair, 2019-03-04:
Tried Netfabb online repair tool. Still same problem

VanessaE, 2019-03-04:
@lordofhyphens I've had this happen before as well on one or two models that I know for a fact were properly manifold.

supermerill, 2019-03-05:
And with the 1.3.0 version?