slic3r/Slic3r #4756: Avoid crossing perimeters is strange

RaabenFuchs, 2019-03-09:
Hello everyone, i'm joining this issue, because i print a power supply case at the moment with **vertical ventilation holes-> many islands**

this causes **ridiculous travel moves (every island one time across the whole surface, half arround the whole objec**t, and then a y axis move across the wall) arround the whole object!
(win 64 v1.3.0)

#3019 #3444 #4412 
# #To add, normally it works.
and as this is issue is so old, may i ask, is it applicable to submit own suggestions of sourccecode?

hope you can estimate the moves a bit, really funny:

[image]

i'm total newby, but really prefer  slicer over cura! thx a lot

lordofhyphens, 2019-03-09:
It's always acceptable to submit suggestions in the form of source code (it's a pull request). You will need to define the behavior though and write tests for it.

RaabenFuchs, 2019-03-10:
hmmm, sounds hard to build this whole test environment. i think rectangular things the feature works.

is this "take a travel around the island" thing another issue?
maybe it travels, even if it should work inside the island

so many people working on slicer, all in their freetime? does anybody work on which corner he wants
looks like many many working grounds

lordofhyphens, 2019-03-10:
The first part of writing a test is figuring out what behavior is desirable. Then you write it up behaviorally:

Given some 3d model geometry at a specific layer,
When the hotend does a travel move, 
Then the travel path looks like (...).

By clearly defining what we think should happen (documenting that with tests), we make it harder to break things in Slic3r.

RaabenFuchs, 2019-03-10:
Ah ok. Very nice,
ill give it a try, building with codeblocks if thats possible. Never compilated a huge project... thx for the hint, i also saw the documentation, that tests are allready included...
But you dont know recently, who ever worked on the mentioned features? im sorry i also have no irc and stuff like that

lordofhyphens, 2019-03-11:
The C++ test suite builds with CMake and uses the CATCH2 framework.

7eggert, 2019-03-19:
This seems to be a good place to ask: "Given some 3d model geometry at a specific layer" - is there documentation about that?

RaabenFuchs, 2019-03-25:
lol. yes, however the travelling over holes seems to be a more common problem.
but this here is kind a interesting, isnt it more a structural thing, different things affecting each other?

thanks anyway @lordofhyphens but im giving it up, trying to build with cmake, dont even now about pearl and wich wx version is used. maybe ill try later under linux but now i first want to print some fancy stuff... bye

closed 2019-03-25 by RaabenFuchs