slic3r/Slic3r #4758: Copy the Estimated printing time result to the Print Summary screen

thiagopeixoto16, 2019-03-11:
Currently this information is only being displayed at **Controller tab** when printing via USB (queue frame).
My suggestion is to copy that field to the **Print Summary** screen for those don't print via USB:
[image]

leonplust, 2019-03-14:
Is anyone working on this already? if no, I would start to work on this.

thiagopeixoto16, 2019-03-15:
@leonplust I'm not working on this. 
...because I'm a humble poor user :)

shepherdsnowdev, 2019-03-15:
+1 request for this!

supermerill, 2019-03-15:
+1, you can start working on it.

lordofhyphens, 2019-05-01:
Thank you to @l29ah for knocking this out.

closed 2019-05-01 by lordofhyphens

thiagopeixoto16, 2019-05-03:
@l29ah Thank you very much:
[image]