slic3r/Slic3r #4761: Next solid layer after Aligned rectilinear angle infill should be perpendicular (internal bridges)

thiagopeixoto16, 2019-03-16:
When the next solid layer after Aligned rectilinear infill is on the same direction it makes the bridges unsupported. See the case below:

Last infill layer:
[image]

Next solid layer:
[image]

In this case it will create sags on solid layer since the bridges aren't supported.

The solution I can imagine is Slic3r to always plan the next solid layer perpendicular to the last infill layer if using  Aligned rectilinear.

For whom face the same problem:
---

As workaround you can change the top layers to match the perpendicular direction. However it don't work if you have multiple top layers in you model because one of then can match the parallel alignment.

prim3d, 2019-06-06:
Hello! I would like to know, how can infill be orientated in one direction? I need such a infill for one project.
Thanks!

CCMCAGP, 2019-08-20:
@thiagopeixoto16 Hi, in a similar way as @prim3d I am looking for an true unidirectional infill. However when I use Aligned Rectilinear, 99%, I still have the cross-hatch pattern for some reason, which renders it redundant with the Rectilinear one. I am also wondering why it is impossible to use 100% with this method. It appears from your picture that you managed to use it as an unidirectional infill at least. Can I ask what version you were using and how did you do this? 
Thanks in advance!

hellraiserinchief, 2019-11-01:
Hi, can someone guide me on how to approach resolving this issue in the code. I see that we have all the fills in xs/src/libslic3r/Fill, but do not know how they connect to the actual infills/top fills

supermerill, 2019-11-02:
The solid fill switch direction every layer. If you want to have a infill with the same direction as solid infill, it's inevitable.

Only solution is to have the infill or solid fill rotated by 45°. To do that, look at the "angle" parameter inside the Fill class that AlignedRectilinear extends.

foreachthing, 2019-11-04:
@supermerill that cheat might work if you only have one object on the platter. If you have two objects (or even a single object) and they require that setting on different layer heights? Well, you'd be attached to an object by an inclined plane, wrapped helically around its axis.

supermerill, 2019-11-04:
@foreachthing 
if sparse infill is switched 45° vs solid infill, and each layer switch the direction by 90° for sparse & solid, sparse will never be aligned against solid.
Or is there something i forgot to think about?

hellraiserinchief, 2019-11-04:
Is there a way to access last layer's directions?

supermerill, 2019-11-04:
basically layer_id%2 ? 0 : 90
see Fill::_infill_direction for more information