slic3r/Slic3r #4762: [Feature request] Hollow support / less dense support(low priority)

7eggert, 2019-03-19:
When I made a model needing a lot of support, I put a box below the most part:

[image]

The box, being set to one shell, no infill, IIRC two or three tops; did consume less material than the original support, also it stuck to the heatbed (while the flimsy support didn't.

Therefore I'd like to suggest a similar feature: If a support structure is larger than a given area and if it's higher than a given treshold, a bridging layer might be introduced to the support and the support below that layer may use wider distances than the default support.

RaabenFuchs, 2019-04-14:
i would appreciate this!
nice idea.

GigaTech, 2019-04-18:
Option to turn off support 'shell' would be great.

Also adjust distances from model, both the nearest wall and furthest wall of the support material from the model.

Thanks