slic3r/Slic3r #4763: Build for distribution Ubuntu 16.04

marcel2034, 2019-03-20:
Hello ! 

I can't seem to find a guide to build the sources for distribution on ubuntu. 

My situation is the following : 
I have a computer with slic3r sources and the admin power to install all the necessary packages and build, 
and another one where I don't have authorizations. I want to launch my slic3r version on that computer. 

I tried to build it, zip it and send it to the other computer, but the perl libraries can't seem to work... 

Is there any wiki that I missed on how to send a distribution version of slic3r ? 

Thanks ! 

**Ubuntu 16.04**

lordofhyphens, 2019-03-20:
Just use the AppImage.

lordofhyphens, 2019-03-21:
You can examine the existing build system and scripts for making AppImages if you want to package your own, but we don't generally support people making distributions of their own.

closed 2019-03-20 by lordofhyphens

marcel2034, 2019-03-21:
Thanks for the answer !