slic3r/Slic3r #4764: Slic3r can't parse the simplest of the conditional code

RuiCarneiro, 2019-03-22:
**Describe the bug**
Since Slic3r lacks the menu option, I'm trying to use conditional Gcode to set the flow rate, it simply can't parse this

**To Reproduce**
After layer change (or before layer change gcode)

    {if [layer_num] == 0} M221 S115
    {if [layer_num] == 1} M221 S100

**Desktop (please complete the following information):**
 - Version latest dev build

lordofhyphens, 2019-03-22:
Please note the git short hash so I have a reference point. "Latest dev" means nothing

RuiCarneiro, 2019-03-23:
> Please note the git short hash so I have a reference point. "Latest dev" means nothing

Done.

lordofhyphens, 2019-03-24:
@RuiCarneiro That's not a hash I recognize (it doesn't show up in the commit history of Slic3r at all).
Current newest build as of writing is https://dl.slic3r.org/dev/mac/slic3r-1.3.1-dev-05a2f90-osx.dmg

I just checked on 05a2f9028 and it works as expected with your provided gcode in the before layer change gcode.

I don't know what's going on.

lordofhyphens, 2019-03-24:
Since I can't reproduce on a current dev build, closing.

closed 2019-03-24 by lordofhyphens