slic3r/Slic3r #4768: IDEA: thicker brim with groove

kwis2, 2019-03-28:
I am wondering if would it be posible to generate brim with a groove to easier brim remove. Idea is to increase layer number of whole brim except small distance from model. On picture there is solution I often use. 
[image]
green it is model, blue color it is single layer brim, and gray is double (multiple) layer brim, which make brim rigid and easy to remove. Sometimes I add small rods in corners which increase adhesion.

wheemzy1, 2019-04-03:
I literally came on just  now to post this idea after my brim was Destroyed and hanging on for dear life for 3 hours.

supermerill, 2019-04-03:
did you try to activate brim & skirt, setting the skirt distance to 0, and increasing the skirt height?

kwis2, 2019-04-03:
@supermerill it could work only with simple model with only "external" walls. If there is any irregularity in model skirt moves away. The same problems if you would like to print multiple obiects.