slic3r/Slic3r #4773: svg ouptut misses very small features

DannyYCCheng, 2019-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

We are using Slic3r for very small prints (with accuracy down to micrometres). We use Slic3r to output a svg file, which is then converted to the format we need. One of the test model has a hole 0.02mm in diameter, when I output this to svg, the holes do not show up. I then scaled the model up 5 times, and all except the 0.02mm (now 0.02*5=0.1mm) holes are visible, the next hole is 0.03mm (scaled up to 0.15mm) so I know holes <=0.1mm aren't visible when outputted to svg, and holes somewhere between 0.1mm to 0.15mm and bigger are visible. The documentation for svg says:

> The majority of options in Slic3r are not required when generating SVG, however the Layer height setting will dictate the number of layers. Note that Slic3r restricts the layer height to be smaller than the nozzle diameter, so this may also have to increased if higher layers are desired.

Unsure if there is a setting in Slic3r that I can change to increase the minimum resolution for svg output?

3D model of the original size and upscaled so 5x the original size attached: [slic3r_resolution_test.zip](https://github.com/slic3r/Slic3r/files/3042231/slic3r_resolution_test.zip)

One layer at original size: [image]

One layer at 5x size: [image]

Configuration: [config.ini.zip](https://github.com/slic3r/Slic3r/files/3042256/config.ini.zip)

OS: Linux
Version: 1.3.0

lordofhyphens, 2019-04-04:
I wonder if the same small feature removal optimizer is still present in
the SVG mode.

For FDM we remove most details smaller than the nozzle (because we can't
print something that size).

If you set the nozzle diameter really small (adjust layer height) as a test
you may see the features in your output.

DannyYCCheng, 2019-04-04:
Thanks @lordofhyphens will give it a try when I get in the office. Our problem is our layer height is usually bigger than our minimum hole diameter, so if this works it'll create another problem... Maybe I can remove that nozzle>layer height restriction in the source code if I find where it is.

lordofhyphens, 2019-04-04:
Yeah that is why I mentioned it as a test. If it works then we can see what
we can do to relax that restriction.

DannyYCCheng, 2019-04-04:
Tried setting:
nozzle_diameter = 0.001
layer_height = 0.005

and (to satisfy the constraint nozzle > height)
nozzle_diameter = 0.01
layer_height = 0.005 

Neither worked.

lordofhyphens, 2019-04-04:
Ok, thanks

DannyYCCheng, 2019-04-04:
In the mean time what I might do is scale up the svg output, then scale down when converting to our format so the other guys can get on with their work (there is a scaling involved from svg to our format anyway so not a biggie). I'll probably be going through the code today to see if I can find where the issue is.

ledvinap, 2019-04-05:
@DannyYCCheng : There are 'negligible size' assumptions scattered all over the code (for example negative offset followed by positive one to remove various artefacts). For your use case it would be probably best to scale model so that smallest important feature is comparable to nozzle size (0.1 mm should do).

supermerill, 2019-04-05:
I wonder if this doesn't come from the TriangleMesh safety_offset (it removes all gaps lower than 0.1mm) . Slic3rPE exported it as a modifiable setting, maybe it's something to consider.

ledvinap, 2019-04-05:
@supermerill : Not sure about current code, but it was hardwired in much more cases in past ...

supermerill, 2019-04-05:
That's almost the only one that expand before retracting. Others are doing the opposite.

marcus-sonestedt, 2024-03-05:
Btw, for SVG it was pretty easy to upscale 10x before slicing, then process the XML to downscale again afterwards, before doing printing.