slic3r/Slic3r #4775: jerk request!

RaabenFuchs, 2019-04-07:
hello again, i spent 2 whole days speeding up to 200 mm/s until i noticed, i missed speeding up the jerk to 1/3 as tolf by some youtube guy (even though i already played around with it in cura)
its just hard to figure out, that you have to play around on the printer, despite it seems to be a common feature. and it seems to be resetet automaically...

or is there some kind of cheatcode to enter expert mode?
well finally i also was confused because i think overhang bridging slowed down everything, too. even as i turned it of?! is that possible?

im a really happy person now, anyway, so tell me, as i know, sb else requested this before, whats the deal

lordofhyphens, 2019-04-13:
Jerk is one of those "set and save to EEPROM" settings. As such, we normally wouldn't set it via slic3r.

Acceleration is controlled via Print Settings (or can be disabled). 

I otherwise have no idea what you're asking for.

RaabenFuchs, 2019-04-14:
despite my firmware doesnt save to eeprom (ok i have to update marlin and dont know how other printers behave)

i thought it could be usefull, tuning or autocalculating this for different speed, because if im going faster (ok 200 maybe was a dream) but say 100, you have to set up tempereture getting more fluidity, but then you have to startquicker, to avoid blobs, and it reduces printing time.

then when you go slower to print details and smaller perimeters on another piece, it might be usefull to auto turn it down again 20 jerk. or to avoid crashes.

think about it

ok i dont want to anoy you, and its on you, but i dont know how many people had asked for it. and of course its somehow an advanced setting, wich could lead to, your hardware wearing out faster.

on this script thing, i tried to change your example code, with the fan speed, but i guess the -fixed -w options didnt work for me. somehow it didnt change my files at all.
normaly i dont use cmd/python. im not sure what to do.

and btw: when i delete the path to the postprocessing script in "print settings" slicer crashes -> im not able to disable the script completely :)     (i open another issue now on that)

thx for your response

RaabenFuchs, 2019-05-15:
Sorry for the unconvinience, finally managed it, through custom g-code, very easy.

**anyway i think jerk somehow limits acceleration** if set to 20 by default. i came out setting jerk:
-to 20 for 1000 accel
-or 30 for 2000 upwards
otherwise it slows down things, but you dont notice as user.

closed 2019-05-15 by RaabenFuchs