slic3r/Slic3r #4776: Skirt Won't go away.

wheemzy1, 2019-04-08:
Does anyone know why the skirt will not disappear? I have tried changing every setting I can think of but nothing is working. Any ideas? Version 1.3.0/Ubuntu 16.04

Thanks

supermerill, 2019-04-08:
print option->skirt->loop : change from 1 to 0
works for me.

wheemzy1, 2019-04-09:
Changing the settings does not work. If I set the skirt to 0 loops it still persists. I then went and installed version 1.2.9 and it did the same thing.

supermerill, 2019-04-09:
I'm sorry, I understood that you only tried to change "every setting I can think of", and was wondering if you maybe forgot this one. That said, i can't help you much more with the amount of information you provided (basically "it doesn't work!"), as i can't reproduce the bug.

- this bug can only appear on your pc -> nobody can help you.
- this bug can only appear on ubuntu 16.04 -> maybe someone with this distribution can help you, but I doubt it's the main cause.
- other -> no enough information

You can now wait for someone one else to help you, or maybe give us more information, like "upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used" as the bug report template say. I strongly encourage you to also join the generated gcode. You have to zip them to be able to drag-drop files on github.

lordofhyphens, 2019-07-07:
@wheemzy1 Waiting on more information. Closing for now as it hasn't been able to be reproduced yet. 

Add the requested information and I will reopen.

closed 2019-07-07 by lordofhyphens