slic3r/Slic3r #4777: End of line on top solid infill and perimeters

xxsteven69xx, 2019-04-09:
I would like to propose an option to hide end of a drawn line to be hidden under previous layer.

[image]
[image]

It would conseal not only these tiny strings but also the small ring the nozzel leaves imprinted on the part.

supermerill, 2019-04-09:
I have a hard time to understand, as there are sometimes hundred of moves per layer. How do you imagine your proposal to work? What if an extrusion began & end "not under the next upper layer"?

Did you try an alternative like disabling z-hop?

xxsteven69xx, 2019-04-09:
Disabling z hop might induce other artifacts like scarring the top surface or catching on a curled edge. For perimeters maybe under seam alignment place seam under the area that's most covered in the next layer. As for top layer solid infill a"quick and dirty way might be too end the fill towards the center of the build plate. A better solution would be to process the layer above and make sure the end of the extrusion occurs under the next layer inside the shell. 
I imagine it would be like the option to not cross perimeters. A checkbox that would warn the user that slicing will take longer and print times would take longer.

xxsteven69xx, 2019-04-09:
To be there I have little to no programming skills. I programmed in basic on my TI-89 graphing calculator in high school, and and the occasional batch or bash script about twice a year since then. If this feature is not possible as it would require a rework of slic3r from the ground up, then that is fine. I'm just throwing ideas out there and seeing what sticks.

xxsteven69xx, 2019-04-09:
example: instead of seam here
[image]

place the end of the extrusion here
[image]

supermerill, 2019-04-10:
Here i also highlight the bit that are gapfill:
[image]

These can't be moved, and it's maybe the bits from them you see in your real piece?

xxsteven69xx, 2019-04-10:
Something I have to thought of