slic3r/Slic3r #4778: Supporting the first bottom solid layer

xxsteven69xx, 2019-04-09:
Let me start off saying I have no experience programming. I wonder how hard it would be to implement a way to guarantee that the first bottom solid layer will allways attach to infill/perimeter.

Example of problem:

[image]

Idea 1 is to take the perimeter and grow/expand it until something like (first_bottom_solid_layer_contact_area)=(first_bottom_solid_layer_perimeter_length) x 50%

[image]

Idea 2 is to just make sure it is attached on all ends

[image]

supermerill, 2019-04-09:
Somethign like that?
[image]

I'm doing something very similar in my fork in https://github.com/supermerill/Slic3r/wiki/Dense-infill-under-solid-layer

> how hard

very

xxsteven69xx, 2019-04-09:
Yeah, that would work. Just trying brain storm on how to make slic3r more idiot proof. I'm a fan of less infill and more perimeters. It has gotten me into trouble a few times. 

[image]
[image]