slic3r/Slic3r #4783: Trying to load invalid config bundle causes crash

tkircher, 2019-04-12:
**Version**
slic3r-1.3.1-dev-5361240-osx.dmg

**Description**
Trying to load an improperly formed config bundle causes an immediate crash

**Operating system**
MacOS 10.12.6

**File in question**
[0.4.4.txt](https://github.com/slic3r/Slic3r/files/3071126/0.4.4.txt)

lordofhyphens, 2019-04-12:
@tkircher I just merged #4765 which may address this.

lordofhyphens, 2019-04-12:
At least with the crash, anyway.

tkircher, 2019-04-12:
I'll check it out and report back. Thanks!

tkircher, 2019-04-12:
Still crashes with slic3r-1.3.1-dev-5f68ce4-osx.dmg.

lordofhyphens, 2019-04-13:
Ah, that's probably because you are trying to load a Prusa3d config bundle and the config loader got more strict about what it accepts (and we hadn't done the error handling to adapt).

lordofhyphens, 2019-04-13:
I am not experienced enough in Perl to handle that error.

tkircher, 2019-04-13:
> Ah, that's probably because you are trying to load a Prusa3d config bundle and the config loader got more strict about what it accepts (and we hadn't done the error handling to adapt).

Even if I'm trying to load a PNG accidentally, the parser either accepts or rejects. Crashing is always a bug. ;)