slic3r/Slic3r #4788: Please offer layer thickness as option for layer count functions ie. top/bottom thickness

awdemuth, 2019-04-17:
I'm a frequent user of the adaptive slicing setting, but problems arise when slicing options are defined by numbers of layers. If I want a thicker top or bottom shell, it's hard to determine the actual printed thickness if adaptive slicing is enabled - areas with thin layers are much weaker than those with thicker layer heights. I've also noticed that if you want to limit support material to a certain height, you can only define this by numbers of layers - this is very difficult to ascertain accurately.  Another area this could be temperamental is with "combine infill layers" or "solid infill layers" options - though this is something I don't have any experience with and cant offer real examples of it's behavior.

It would be nice to define a certain height in mm for these features, that way one can have their print turn out as expected if they change their adaptive slicing settings. 

For support height limitations, I've briefly considered calculating out how many layers I'd need based off of rough layer estimates, but ended up going the trial end error route, which is fine for small models, but sometimes the model is quite complicated and slicing time takes quite a while. I grew tired of this method of these models, considering the inaccurate results it gives and just went with fixed layer height, giving up the advantages of adaptive slicing.

For Top/bottom layers, I usually just add a few layers, but this is wasteful, as these two layers are the most time/resource intensive layers. I'm unsure how this affects overall wall thickness. on sloped areas. 

Thanks!

lordofhyphens, 2019-04-17:
See "minimum shell thickness"

awdemuth, 2019-04-17:
After going through (what I think is) every option under every view, I can't seem to find a minimum shell thickness value to change. Is it in the slic3r 1.3.0 Windows GUI?  A search brings mention of it in config files, but nothing in the GUI

This is a valid solution to the first of my issues, but limiting support material height is still a concern of mine, and combine infill layers/ solid infill layers still remains unaddressed.

lordofhyphens, 2019-04-17:
[image]
1.3.1-dev.

awdemuth, 2019-04-17:
Awesome, I just needed to switch over to the development version.

closed 2019-04-28 by lordofhyphens