slic3r/Slic3r #4789: Minimum Time by Layer

wjallison1122, 2019-04-17:
**Is your feature request related to a problem? Please describe.**
I have an issue with my prints where vertically thin features (layer area of less than .125in^2) turn out 'blobby' or 'melted,' for example when a part comes to a point.
Example:
https://i.imgur.com/aGtSwmH.jpg
I (and others who I've asked) believe that this is caused by the printer moving on to the next layer before the prior layer has had time to cool.  

"Slow down if layer print time below" has proven to be ineffective.  

**Describe the solution you'd like**
The difficult way to solve this would be to make Slic3r track how long it's been since the print head passed a particular (x,y), and set a minimum time until it's allowed to pass the area above that coordinate again.  This could take the form of switching order of perimeters to infill, changing the direction of perimeter / infill print, or, if no other options are available, moving the print head away.  
The easy way is to add a way for the user to specify a minimum amount of time that a layer can take.  If a layer takes less than that amount of time, the printer will retract and wait until the time is up before continuing with the next layer.  
I recognize that this has been considered and rejected in the form of "orbit," but something needs to take its place.  

**Describe alternatives you've considered**
Alternatives would be slower movement for layers with a small area or increased fan speed for layers with small area.  Fans are already at max, so the latter isn't really an option.

VanessaE, 2019-04-17:
See also #4006

AbeFM, 2019-05-22:
Not the exact same thing, but I've seen several other slices which lift the head and set it aside when they need more time between layers. It takes a lot less math to do this, and it works well - though it does nothing for nozzle pressure, etc.

VanessaE, 2019-05-22:
That's why I recommend a prime/dwell tower.

AbeFM, 2019-05-22:
Not against it, I think you're right.

BUT.... it's a LOT easier to implement "lift, wait, unlift", in terms of
reward on effort spent that seems a no brainer.

VanessaE, 2019-05-22:
Not when you factor-in the notion that you'll have to do something to contain the ooze that will happen from dwelling more than a second or two, and to bring the pressure back up before the nozzle gets to any part of the object that can show through to the outside.

AbeFM, 2019-05-24:
Not arguing, all I'm saying is in practice, I've used a simple lift 100's
of times, and got great prints. It probable triggered a retraction when I
did it.