slic3r/Slic3r #4791: Improved Arrange algorithm

sanchosk, 2019-04-23:
**Is your feature request related to a problem? Please describe.**
For me this is a bit of a problem, but I believe in general this is more a feature request.
When arranging more objects, Slic3r tends to "spread" them to width rather than uniformly.
As an example - add 1 small object, make 3 more copies and arrange. I would expect a grid of 2x2, instead Slic3r ends with 1 row with 3 object copies and 1 object copy in 2nd row.

**Describe the solution you'd like**
I would expect the arrangement to end up as a 2x2 group, minimizing the bed usage and travel distance for print head.

**Describe alternatives you've considered**
I usually manually re-arrange the objects.

**Additional context**
I am providing the screenshot with arranged parts using the function and what I would expect as the algorithm behavior.
I am using 1.3.1-dev version with fresh pull this morning.
[image]
[image]

VanessaE, 2019-04-23:
This one offends my OCD, too :smiley:

lordofhyphens, 2019-04-28:
The auto arrange is getting some love from GSoC this year.