slic3r/Slic3r #4792: slic3r not rendering correctly this file

swolter1975, 2019-04-24:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Slic3r is not rendering correctly a file stl

**To Reproduce**
Steps to reproduce the behavior:
1. Go to file-open and ipen the attached file i send
[UpperArm - Finger slider-export.zip](https://github.com/slic3r/Slic3r/files/3112698/UpperArm.-.Finger.slider-export.zip)
[This side of the STl file is not being rendered properly.docx](https://github.com/slic3r/Slic3r/files/3112703/This.side.of.the.STl.file.is.not.being.rendered.properly.docx)

2. Click on '....'
3. Scroll down to '....'
4. See error

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

VanessaE, 2019-04-24:
Your report is incomplete.

Please embed screenshots (not as a proprietary document), add OS and Slic3r release info, and your config.

swolter1975, 2019-04-24:
i have updated the report please review it thank you.

VanessaE, 2019-04-24:
Please add the STL file.

swolter1975, 2019-04-24:
added as zip file thanks

foreachthing, 2019-04-24:
Your STL is faulty.
Use https://service.netfabb.com/ to fix it.

[image]
Selected: fixed. Yours in cyan, the faulty one.

Also, what CAD have you used to draw this? Or, try to increase accuracy with your STL-export:
[image]

VanessaE, 2019-04-24:
Good eye, @foreachthing.

swolter1975, 2019-04-24:
Gilbert

thanks for your reply. its weird because i opened the STL file on idea
maker, also in cura and in both it opened well, the only program where it
does not load well was slic3r.
so n not sure 100% what the probem weas. i did not produce the stl file i
just downloaded from the web.

thank you anyways

rgds,

sw

closed 2019-04-25 by VanessaE

lordofhyphens, 2019-04-25:
@swolter1975 that other programs open a garbage file doesn't mean much if the file is indeed garbage, just that their edge cases work out differently than us.