slic3r/Slic3r #4793: Failure to fill a perforated surface

l29ah, 2019-04-25:
**Describe the bug**

[image]

**3D Model and Slic3r Configuration Export**
[bug.zip](https://github.com/slic3r/Slic3r/files/3117484/bug.zip)

**Expected behavior**
Continuous surface.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 85dd6633fbe186cb4872c944ca54f49402634c57

supermerill, 2019-04-25:
I think the space between the holes is too small for 2 perimeters and the gap fill is lost.

anyway, i don't see how you can print that. My advice is to cut it just above the holes, and flip the lower part by 180°, and glue the two part together after printing.

For you hole problem, you should deactivate the thin wall, and increase the external perimeter width until holes are touching themselves.

l29ah, 2019-04-25:
I would print it by pouring plastic in a three-directional pattern that produces hexagonal holes.
Not sure what do you mean. I've set vertical shells to 0, but no effect, while `Detect thin walls` checkbox is greyed out.

supermerill, 2019-04-25:
[buchner-funnel_better.zip](https://github.com/slic3r/Slic3r/files/3118635/buchner-funnel_better.zip)