slic3r/Slic3r #4794: Print Server Upload - HTTPS not supported

PrestonL, 2019-04-26:
**Describe the bug**
Slic3r is not able to connect to Octoprint using HTTPS

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printer Settings and enter the Hostname and API key of an octoprint server
2. Click on "test"
3. See error:

> I wasn't able to connect to OctoPrint (501 Protocol scheme 'https' is not supported (LWP::Protocol::https not installed)). Check hostname and OctoPrint version (at least 1.1.0 is required).

**Expected behavior**
Slic3r should be able to use HTTPS to communicate with Octoprint

**Desktop (please complete the following information):**
 - OS: macOS 10.14.4
 - Version 1.3.1-dev-85dd663

lordofhyphens, 2019-04-26:
#3718 

I've been avoiding adding SSL/HTTPS support because it adds yet another Perl dependency to the build environment for publishing.

lordofhyphens, 2019-04-26:
Dupe of  #3203

lordofhyphens, 2019-04-26:
should pop up over at https://dl.slic3r.org/dev/mac/branches/ PR4795 when it's done

PrestonL, 2019-04-27:
Just downloaded and ran PR4795, and am still receiving the same error:

[image]

Anything else I need to do?

lordofhyphens, 2019-04-27:
@PrestonL LWP::Protocol::https  should be installed; I've tried to install something different.

lordofhyphens, 2019-04-27:
@PrestonL #tryitnow

PrestonL, 2019-04-27:
Which build will that be on? I'm only seeing "slic3r-1.3.1-dev-f488cc9-osx-PR4795.dmg" on the automatic development builds, and it's unchanged from the one above.

lordofhyphens, 2019-04-27:
[image]

Looks like it should be there, looking at f488cc9 build

PrestonL, 2019-04-27:
I see https.pm in the same folder on the same build, but I still receive the same error trying to connect to an OctoPrint server with SSL enabled.

lordofhyphens, 2019-04-27:
Ok, I don't currently have a solution then for you.

lordofhyphens, 2019-04-28:
@PrestonL I've got one more thing to try; merged in some work by @Salandora.

PrestonL, 2019-04-28:
That did result in a different error message when using the "Test" button.

[image]

"Sent to Printer" results in a different error message:

[image]

If I add an https:// prefix to the address and press "Test" or "Send to Printer" I get this error (it reads the port as :80 even if I add :443):

[image]