slic3r/Slic3r #4798: Slic3r AppImage does not start on Fedora 30

simonCor, 2019-05-01:
**Describe the bug**
Slic3r AppImage does not start on Fedora 30 (It worked good on Fedora 29). The error is:
`/tmp/.mount_Slic3rnFAFBM/usr/bin/perl-local: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory`

I was unable to find libcrypt in the Fedora 30 repositories.

I tested the following development release: Slic3r-1.3.1-dev-2333203-x86_64.AppImage

**To Reproduce**
1. Install a fresh Fedora 30 (I updated from 29 to 30)
2. Start Slic3r ./Slic3r-1.3.1-dev-2333203-x86_64.AppImage

**3D Model and Slic3r Configuration Export**
na

**Expected behavior**
Slic3r should start.

**Screenshots**
na

**Desktop (please complete the following information):**
 - OS: Fedora 30
 - Version: Slic3r-1.3.1-dev-2333203-x86_64.AppImage

**Additional context**
na

lordofhyphens, 2019-05-01:
@probonopd Any suggestions?

lordofhyphens, 2019-05-01:
https://pagure.io/fedora-docs/release-notes/c/01d74b33564faa42959c035e1eee286940e9170e?branch=f28

@simonCor try symlinking libxcrypt.so.1 to libcrypt.so; if it starts then we know what's probably going on.

simonCor, 2019-05-01:
When i do ls -fl libcrypt* in /usr/lib64 i get:
```
lrwxrwxrwx. 1 root root      17 15. Mär 12:29 libcrypt.so -> libcrypt.so.2.0.0*
lrwxrwxrwx. 1 root root      17 15. Mär 12:29 libcrypt.so.2 -> libcrypt.so.2.0.0*
-rwxr-xr-x. 1 root root  257648 15. Mär 12:29 libcrypt.so.2.0.0*
-rwxr-xr-x. 1 root root 3642480  8. Feb 09:02 libcrypto.so.1.0.2o*
lrwxrwxrwx. 1 root root      19 16. Apr 15:07 libcrypto.so.1.1 -> libcrypto.so.1.1.1b*
-rwxr-xr-x. 1 root root 4232912 16. Apr 15:07 libcrypto.so.1.1.1b*
lrwxrwxrwx. 1 root root      19  8. Feb 09:02 libcrypto.so.10 -> libcrypto.so.1.0.2o*
lrwxrwxrwx. 1 root root      23  4. Apr 12:45 libcryptsetup.so.12 -> libcryptsetup.so.12.4.0*
-rwxr-xr-x. 1 root root  461256  4. Apr 12:45 libcryptsetup.so.12.4.0*
```
So there is already a symlink libcrypt.so

I tried the following:
`sudo ln -s libxcrypt.so libcrypt.so.1
`
Then i get the following error when executing Slic3r:
```
./Slic3r-1.3.1-dev-2333203-x86_64.AppImage 
/tmp/.mount_Slic3rkCiGbf/usr/bin/perl-local: /lib64/libcrypt.so.1: version `GLIBC_2.2.5' not found (required by /tmp/.mount_Slic3rkCiGbf/usr/bin/perl-local)
```

probonopd, 2019-05-01:
They did it again. They messed around with glibc. Argh. Maybe we need to bundle `libcrypt.so.1` now, or change perl-local so that it does not use it.

lordofhyphens, 2019-05-02:
@probonopd I may have read your first couple sentences in Charlton Heston's voice from Planet of the Apes.

probonopd, 2019-05-02:
Looking into it. For now I would say let's test what happens when `libcrypt.so.1` from the build system is bundled inside the AppImage like the other libraries. Will be looking into this myself, too.

lordofhyphens, 2019-05-03:
@simonCor there should be a PR build showing up in the branches dir soonish.

simonCor, 2019-05-03:
@lordofhyphens ./Slic3r-1.3.1-dev-041b9b5-PR4800-x86_64.AppImage starts without error. Thank you!

probonopd, 2019-05-03:
Works for me on Ubuntu 18.04, too. :+1:

PR #4800 (2019-05-04): Add libcrypt.so.1 to appimage.
Probably fixes #4798
closed 2019-05-04 by lordofhyphens