slic3r/Slic3r #4799: not obeying layers height

prchal, 2019-05-02:
Slicer doesn't obey layer settings when printer Z full step/mm is used.

**Steps to reproduce the behavior:**
1. set printer's Z full step/mm to 25
2. set extruder's limits min to 0.16 and max 0.2 (layer height)
3. set printing layer height to 0.2/0.2, no adaptive slicing 
4. sliced some layers 0.16, some 0.24, for example layer 9.76 and next 10.0

**Expected behavior**
Fixed layers should be fixed.
Not exceeded max layer height.

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu, Manjaro
 - Version: 1.3.0
[3d.zip](https://github.com/slic3r/Slic3r/files/3137924/3d.zip)

dwillmore, 2019-05-08:
Reproduced with Slic3r-1.3.1-dev-85dd663-x86_64.AppImage on Fedora 29 x86_64.

lordofhyphens, 2019-05-09:
So if I understand this situation, you have specified specific layer heights via the UI and would like Slic3r to do exactly that, even though you also told Slic3r to round all layers  to the nearest full Z step?

lordofhyphens, 2019-05-09:
For people wanting to experiment, the `z_full_steps_per_mm` behavior is in `PrintObject::adjust_layer_height()` (in PrintObject.cpp:547 or thereabouts). It is called by `PrintObject::generate_object_layers()` 

The minimum and maximum layer heights are defined separately for all extruders, but `generate_object_layers` will calculate conservative boundaries, so pass those into `adjust_layer_height`

prchal, 2019-05-09:
> So if I understand this situation, you have specified specific layer heights via the UI and would like Slic3r to do exactly that, even though you also told Slic3r to round all layers to the nearest full Z step?

The specific layer heights which I set are rounded to Z full step. 0.2/(1/25) = 5 full steps. I wanted only this height 0.2, but slic produced 0.24, although it's rounded to full step, it's more then I wanted (for nozzle dia 0.3).