slic3r/Slic3r #4802: Dutch translation

Tillema, 2019-05-04:
I like to see Slic3r in my native language: Dutch. So, I translated all strings for you (see attached file). Would be very awesome if you integrate this feature in the next version of Slic3r.
[nl_NL.txt](https://github.com/slic3r/Slic3r/files/3144473/nl_NL.txt)

(Had to convert it to TXT file because couldn't upload a PO file)

Of course you can always contact me for new strings of this translation.

lordofhyphens, 2019-05-04:
@Tillema You submit a PO file through a pull request.

Oekn5w, 2019-05-20:
@Tillema I see your master branch has the file, do you want to PR it? Doesn't look like any of the translations are used at the moment, they seem to be placeholders for now. So no further modifications seem necessary at the moment.

Tillema, 2019-05-20:
Hi, thank you for your reaction. I'm new to github, so I don't know how to fix that. My intention was to make the Dutch translation available for my fellow Dutchies. A few of my Dutch and Belgian friends don't speak English very well, so they would like a Dutch translation.

Oekn5w, 2019-05-20:
That's all right, I think we all had our problems understanding the workflow. Basically, only the members have write access here, so we contribute by pushing to our forks and offer our changes via pull requests. Then they are reviewed and accepted or denied / changes requested by the repository owners. There are a couple of guides to this ... [Example 1](https://gist.github.com/MarcDiethelm/7303312), [Example 2](https://github.com/firstcontributions/first-contributions). You are almost done, only missing the pull request.

Tillema, 2019-05-20:
Thank you for your attention! According to me, it's done right now.