slic3r/Slic3r #4805: Slic3r crashs loading STL

ilovelinux, 2019-05-10:
**Describe the bug**
I need to print the gears of this package: https://www.thingiverse.com/thing:974906
When I load the STL file, Slic3r crashes. This is the output:
```
$ slic3r
realloc(): invalid next size
Aborted (core dumped)
```
I also tried to fix the .stl file with `slic3r --repair` but it doesn't work too.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://www.thingiverse.com/thing:974906'
2. Download and unzip files
3. Open _gears.stl_
4. See error

**3D Model and Slic3r Configuration Export**
[issue.zip](https://github.com/slic3r/Slic3r/files/3167614/issue.zip)

**Expected behavior**
Open STL file

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Version 1.3.0 (installed by [AUR](https://aur.archlinux.org/packages/slic3r/))

lordofhyphens, 2019-05-10:
Make sure you pass the model through netfabb or something similar.

lordofhyphens, 2019-05-18:
Interestingly, doesn't crash when running through the CLI.
[gears-gcode.zip](https://github.com/slic3r/Slic3r/files/3193613/gears-gcode.zip)

lordofhyphens, 2019-05-18:
I have a workaround submitted in a pull request so you can see the model in all of its terrible glory.

PR #4811 (2019-05-21): Use admesh's stl_repair() instead of simulating it ourselves.
fixes #4805
commit de538954 (2019-05-21): Use admesh's stl_repair() instead of simulating it ourselves.
fixes #4805
closed 2019-05-21 by lordofhyphens