slic3r/Slic3r #4808: SLic3r console --end-gcode doesn't work as expected

JoaquinTrinanes, 2019-05-17:
**Describe the bug**
Using Slic3r-console version 1.3.0. Executing it with the `--help` option it shows that the parameter `--end-gcode` overwrites the default settings (more specifically, the M104 S0 command). However, slicing a file with this option won't remove them.

**To Reproduce**
Execute
`Slic3r-console.exe --end-gcode end.gcode model.stl`
end.gcode can have any content, however, the resulting file will have extra codes after that.
In this case, it simply contains the line `; my custom end gcode`

**Expected behavior**
The gcode file should end with my custom code.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.3.0

lordofhyphens, 2019-05-17:
Put a M104 command in the end gcode.

JoaquinTrinanes, 2019-05-17:
Well, it's a bit hacky, but adding the codes with the temperatures I intend to keep deletes the codes turning them off. Thanks!

closed 2019-05-17 by JoaquinTrinanes