slic3r/Slic3r #4810: Tired of waiting for the Plater I don't use...

stroodies, 2019-05-17:
Hi Buddies.

 Since the last update in 2018, every time I change a number on my Print Settings, Printer Settings, or Filament Settings in Slic3r the GUI lags for a couple of seconds, is this just me?

I believe the Slic3r is causing the delay by attempting to modify its print preview on the "Plater" tab. I don't ever load models into Slic3r directly, I still use Repetier. So, the Plater is never, ever actually doing anything other than making me wait. Turning off "Background Proccessing" has no affect. The Plater Tab cannot be closed.

Is there a setting I'm missing or could a feature be added to disable the Plater Window and its preview functions? 😘

lordofhyphens, 2019-05-18:
It's not that. I believe it's the config validation code, but I haven't been able to nail down the exact cause.

stroodies, 2019-05-18:
Thanks for the response! I take heart knowing you're on the case 😁👍
--Rick