slic3r/Slic3r #4812: GSoC 2019 - SVGNest in Slicer

lordofhyphens, 2019-05-20:
Development log for GSoC 2019 project to implement #3237.

gege2b, 2019-05-20:
Just FYI, there is a C version of SVGNest from the same author : [Deepnest](https://github.com/Jack000/Deepnest)
I don't know if it could be of any help

3bhady, 2019-05-20:
Thanks @gege2b I'll try run it and test it. 
It would be really helpful to have a C implementation rather than JS :)

3bhady, 2019-05-22:
Today I'm investigating the auto-arrange code in slic3r.

supermerill, 2019-05-22:
There may be two different entry point for auto-arrange: 
* the algo that move them around
* the check that happens each time something is changed in the platter.
Be sure to use the same  check algo for both, or to be sure that the check can't trigger after an arrange.

3bhady, 2019-06-16:
I'm having some trouble finding if the polygon is clockwise or anti-clockwise.

If anyone can help point me to something I can read on that I'd be grateful.

What I understand so far is that clockwise means that it's stored in a way that it's points are going in clockwise direction, but I'm not sure how to calculate that. I'm guessing it has something to do with cross product of every two consecutive vectors or something like that.

Samir55, 2019-06-16:
You can check Polygon.hpp in xs/src/libslic3r as it has is_clockwise function 
https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/Polygon.hpp#L36

3bhady, 2019-06-16:
Thanks @Samir55 :)

luzpaz, 2021-01-30:
Did anything come of this? Asking for the FreeCAD folks who are interested in a similar function for the Path/CAM workbench.

bubnikv, 2021-01-30:
You may want to look into the arrangement in PrusaSlicer, inspired by svgnest.
Cura recently adopted our PrusaSlicer code.

luzpaz, 2021-01-30:
@bubnikv cool, care to share a link ?

bosd, 2021-03-21:
Als interested in a nesting feature for freecad, so came across this issue.
@bubnikv This might be of interest to you:
https://github.com/grotius-cnc/Nest4J
Still java, no c++ but more interfacing features.

bosd, 2021-03-21:
@luzpaz I think this is the link you were looking for to the nest2D library.
https://github.com/tamasmeszaros/libnest2d

Theres also this porting attempt to c++
https://github.com/kallaballa/nestcpp