slic3r/Slic3r #4816: Speed weirdness

kajsilander78, 2019-05-26:
Steps to reproduce the behavior:
1. Set all print speeds to 15mm/s
2. Slice a 20mm small coin, 3mm high
3. Notice the speed increase at 2nd layer after printing perimeters....same thing with top infill.

Dont know if extrusion is scaled to match speed, printed tpu when I noticed this. Got massive underextrusion, but it could be extruder limitation.

[files.zip](https://github.com/slic3r/Slic3r/files/3220942/files.zip)

lordofhyphens, 2019-05-26:
Is there a config export in that zip?

kajsilander78, 2019-05-26:
yep

kajsilander78, 2019-05-26:
[image]

kajsilander78, 2019-05-26:
Tried both nightly and 1.3 and think I found the issue. Bridge speed is 60mm/s! That seems to set top layer speed.....

lordofhyphens, 2019-05-26:
Just looking at the second layer indicates that Slic3r is treating that infill like it's a bridge.

lordofhyphens, 2019-05-26:
Here's 0.1 flow ratio for bridges (which Slic3r will show a preview for)
[image]

Same layer, 1.0:
[image]

kajsilander78, 2019-05-26:
Oh, thought bridges only was for bridges... Wanted some stretch for tpu bridges...and was experimenting with bridgespeed and flow for bridges when I ran into this issue. ;) Tried changing nozzle and calibrating extruder and so on.....

lordofhyphens, 2019-05-26:
It's for anything that would act like a bridge. I don't know why that specific layer is presenting as a bridge though (verbose gcode doesn't indicate it should be).

lordofhyphens, 2019-05-26:
Remember that % in width is over layer height, not default extrusion width.

closed 2023-04-10 by kajsilander78