slic3r/Slic3r #4817: Crashed during slicing

limegreensky, 2019-06-02:
### Version

1.3.0

and

1.3.1 dev (6bb8a75)

### Operating system type + version

MacOS 10.14.5

### Behavior

Infinite time to generate GCODE in 1.3.0

Crashed after several minutes in "infilling layers" state at 1.3.1 dev...

#### STL/Config (.ZIP) where problem occurs

[STL.zip](https://github.com/slic3r/Slic3r/files/3244672/STL.zip)

foreachthing, 2019-08-27:
I've run it through Blender with a DECIMATE modifier, then exported it with the 3D-Printer addon.

There are/were zero- and intersecting faces ... a lot of 'em!

[image]

After that, I was able to slice the stl in Prusa and Slic3r.

[image]

[untitled-WIHA_Secure_Torx_Holder_-_001.amf.zip](https://github.com/slic3r/Slic3r/files/3544531/untitled-WIHA_Secure_Torx_Holder_-_001.amf.zip)

[untitled-WIHA_Secure_Torx_Holder_-_001.zip](https://github.com/slic3r/Slic3r/files/3544515/untitled-WIHA_Secure_Torx_Holder_-_001.zip)

rceric-AZ, 2019-08-27:
The files you sent are not the files I created.  Below are my very simple files.  They were created in SolidWorks 2019 and saved as STL's.  I thought perhaps the shell feature could be causing the problem, but these are version 2.  Version one had no problems when slicing or printing.

Thanks for your help,

Best,

Eric

rceric-AZ, 2019-08-27:
>     Woop! Sorry! :-)
>
>     Well, one issue I see with your model is, that it's too thin (with my settings anyway).
>     But Slic3r/Prusa does not crash ...

You found the problem.  I changed the shell thickness to .05" and it now slices as expected.

Thank you very much for your assistance.

Best,

Eric

foreachthing, 2019-08-27:
Woop! Sorry! :-)

Well, one issue I see with your model is, that it's too thin (with my settings anyway).
But Slic3r/Prusa does not crash ...