slic3r/Slic3r #4818: Provide option to generate fan control (M106) as a percentage duty cycle instead of 0-255

mvadu, 2019-06-04:
**Is your feature request related to a problem? Please describe.**
I am using Smoothieware as my 3d printer platform. In my case I am specifically using a pinn which generates hardware PWM, where the `S` value denotes the duty cycle in percent. In the Slic3r cooling control I can set the max value for the fan control, but it always assumes the `S` value for `M106` to be in a 0-255 scale. 

**Describe the solution you'd like**
Having an option to define (May be at printer level as the flavor of M106 is more board specific i.e. whether board is using software or hardware PWM than the filament specific) will be great for his case.

**Describe alternatives you've considered**
I can manually calculate actual maximum duty cycle allowed (example for a specific filament I want to run the cooling only up to 89%) using max%/255*100 (89/255*100=~35%) and entering that in the filament config section as max percentage. 

**Additional context**
My current solution works, but that makes the filament configuration not portable. I can't share my filament config with anyone who is not using same setup of hardware PWM like mine. Before sharing I have to remember to update the config to 89% in my example.

lordofhyphens, 2019-06-04:
One other solution you can do is add a post-processing script to rewrite the M106 calls in the interim.

For someone looking to add this attribute, see https://github.com/slic3r/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI for adding new items (this should be under [GCodeConfig](https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L325)).

To make it actually do something, assuming you added the option to GCodeConfig, edit [set_fan](https://github.com/slic3r/Slic3r/blob/master/xs/src/libslic3r/GCodeWriter.cpp#L163) to modify the output gcode.

I would probably add another if/else block for Smoothieware gcode flavor and disable the control for the fan max duty cycle in the appropriate [_update](https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/PresetEditor.pm#L1596) method to disable it when it's not applicable.

lordofhyphens, 2019-07-07:
@mvadu So you're looking for an option in the Printer config that converts the S values used to a truncated percentage (normalized between 0-100)?

mvadu, 2019-07-07:
Yes, instead of 0-255, it should generate 0-100 and honor the max and min percentage options from filament cooling sections.

lordofhyphens, 2019-07-07:
Ok, it's currently just scaled in set_fan to 0-255 from the normal percentages. Can you give branch feature-4818 a try and see if it's what you are looking for?

lordofhyphens, 2019-07-07:
https://github.com/slic3r/Slic3r/tree/feature-4818

lordofhyphens, 2019-07-07:
Does smoothie actually need the % signifier?

mvadu, 2019-07-07:
No, it's just `M106 S35` indicating 35% duty cycle.. I will give your branch a try later today and report back. Thank you for considering my suggestion and spending time to implement it.

mvadu, 2019-07-10:
@lordofhyphens is there a PR you created (so I can pull the exe from AppVeyor CE server)? I am afraid I don't have the build toolchain setup for me to build an exe myself.

lordofhyphens, 2019-07-10:
Yeah appveyor has an artifact

mvadu, 2019-07-21:
Hi @lordofhyphens sorry for getting back this late. your tree does not have an artifact on AppVeyor, since you did not create a PR. Can you please create a PR for the latest commit? Looking the code, it should work.

PR #4843 (2019-09-23): Allow fan to be specified from 0-100 (decimal) instead of 0-255
Fixes #4818
closed 2019-09-23 by lordofhyphens