slic3r/Slic3r #4819: Filename Change

cycl0ne, 2019-06-04:
You lately changed the filename output to the time being used. Thats wonderfull and this took me to dig more into the filename thing. So my request is as follow, could you change the default output name to:
{input_filename_base}_{layer_height}mm_{filament_settings_id[0]}_{printer_model}_{print_time}.gcode

filament_settings_id instead of filament_name?

You may ask why? Easy: One has different PLA which is used, with different temperature. If the output file is just: 

name_0.2mm_PLA_MK3S_27m.gcode

its not usable, but when used the filament_settings_id:

name_0.2mm_PRUSA_PLA_MK3S_27m.gcode
name_0.2mm_GENERIC_PLA_MK3S_27m.gcode

its makes more sense.

lordofhyphens, 2019-06-04:
Pull requests move faster :)

See PrintConfig.cpp for the default values.

It wouldn't affect existing config files (which is why we rarely change defaults)

lordofhyphens, 2019-06-04:
See your own Print settings though to set this for your profiles.

cycl0ne, 2019-06-05:
I know :-) But, im lazy. I changed it yesterday in default config in the program directory (didnt want to create 8 own configs for the default prusa configs). but with a newer config update it gets overwritten. As written its just a suggestion from my side, since it makes more sense than just stating the material.

cycl0ne, 2019-06-05:
ah sorry, just noticed im in the wrong issue tracker :-) this has to go to prusa3d/PrusaSlicer-settings. So closing it here. Sorry for the hassle.

closed 2019-06-05 by cycl0ne

lordofhyphens, 2019-06-05:
It's still a legit request for Slic3r.