slic3r/Slic3r #4821: Allow for Extrusion-Width settings as percentage of nozzle diameter

MrMabulous, 2019-06-07:
In the advanced Print settings, it's possible to set extrusion width. However, only in the auto setting the width is derived from nozzle diameter. If a value is set, it's either absolute or expressed in terms of layer height. This means print settings and printer settings are not orthogonal and for each printer setting with altered nozzle diameter, one also needs a new print setting (unless one uses automatically chosen values).

Would be great to be able to express extrusion width in terms of nozzle diameter.

EvilGremlin, 2019-10-19:
Second that, calculation as percentage of layer height makes absolutely no sense.

wavexx, 2019-10-25:
On Fri, Oct 25 2019, Merill wrote:
> It's (a relic?) for printer with constant flow.

Ok - wow, that actually does make sense in that context.

lordofhyphens, 2019-10-25:
Also there was a lot of discussion of what the optimal width over height
was for strength.

Trellian, 2019-10-25:
@lordofhyphens Here's an YT video by Thomas Sanladerer, where he tests the effect on strength of various extrusion width parameters.

It looks like and extrusion width of about 140% of nozzle width is about the optimum, if you are looking for strength.

https://www.youtube.com/watch?v=9YaJ0wSKKHA&t=659s

supermerill, 2019-10-25:
@Trellian it's Stephen from cnc kitchen, totally not Thomas Sanladerer (but both may be German).

Trellian, 2019-10-25:
Whoops, my bad. Age is definitely catching up with me :o