slic3r/Slic3r #4823: Does not start, libGL error: unable to load driver: r600_dri.so

probonopd, 2019-06-11:
Cont'd from https://github.com/slic3r/Slic3r/issues/4663#issuecomment-500927049

This issue seems to be back.

Fails to run on `xubuntu-18.04-desktop-amd64.iso`:

```
me@host:~$ /isodevice/Applications/Slic3r-1.3.0-x86_64.AppImage 
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1192 error_code 2 request_code 156 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

me@host:~$ /isodevice/Applications/Slic3r-1.3.1-dev-48d5b11-x86_64.AppImage 
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 575 error_code 2 request_code 156 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

The PrusaSlicer AppImage is working on the same system.

probonopd, 2019-06-16:
On `xubuntu-19.04-desktop-amd64.iso` `Slic3r-1.3.1-dev-a0a2055-x86_64.AppImage` launches correctly.

JasonB579, 2019-07-01:
I'm getting a similar issue on `xubuntu-18.04-desktop-amd64.iso`. I've tried the most recent official release, `Slic3r-1.3.1-dev-a0a2055-x86_64.AppImage`, and `Slic3r-master-latest.AppImage`.

I don't have an issue missing `r600_dri.so`, but I am missing `swrast_dri.so`.

```
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 570 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

probonopd, 2019-07-01:
@JasonB579 just for informational reasons, can you try whether the PrusaSlicer AppImage runs on your system?

JasonB579, 2019-07-01:
Seems to work without a hitch.

probonopd, 2019-07-01:
OK, so we need to compare which graphics driver libraries they are bundling vs. what Slic3r is bundling, and adjust accordingly.

DaniDiaz71, 2019-08-21:
Hello,

I have similar error but with radeonsi and swrast [Ubuntu 19.04 - Proc: Intel® Core™ i5-2500K CPU @ 3.30GHz × 4 - Graphics: AMD® Tahiti]:

`libGL error: MESA-LOADER: failed to open radeonsi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'GLXBadContext'.
  (Details: serial 1444 error_code 161 request_code 153 minor_code 6)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
`
 If I search those files, I can find several names but not sure how to solve this problem... :(

`dani@Bigboy:/opt/Slic3r$ find / -name "*radeonsi*" 2>/dev/null
/usr/lib/x86_64-linux-gnu/dri/radeonsi_drv_video.so
/usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
/usr/lib/x86_64-linux-gnu/vdpau/libvdpau_radeonsi.so.1
/usr/lib/x86_64-linux-gnu/vdpau/libvdpau_radeonsi.so
/usr/lib/x86_64-linux-gnu/vdpau/libvdpau_radeonsi.so.1.0
/usr/lib/x86_64-linux-gnu/vdpau/libvdpau_radeonsi.so.1.0.0
dani@Bigboy:/opt/Slic3r$ 
dani@Bigboy:/opt/Slic3r$ 
dani@Bigboy:/opt/Slic3r$ 
dani@Bigboy:/opt/Slic3r$ find / -name "*swrast*" 2>/dev/null
/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
/usr/lib/x86_64-linux-gnu/dri/kms_swrast_dri.so
`

Any idea?

Thanks and BR.

DaniDiaz71, 2019-08-24:
Solved using: 

`LD_PRELOAD='/usr/lib/x86_64-linux-gnu/libstdc++.so.6' ./Slic3r-1.3.0-x86_64.AppImage `

probonopd, 2019-09-26:
Please see if the issue disappears when you run

    ./Slic3r-1.3.0-x86_64.AppImage --appimage-extract
    rm -rf squashfs-root/usr/lib/libstdc++.*
    ./squashfs-root/AppRun

If it works using those commands, then the following options might solve this:

* If possible, compile on the oldest still-supported Ubuntu LTS release (currently xenial) without adding compilers (e.g., through ppas) that draw in newer versions of `libstdc++.so.6` than what comes with the distribution. Do _not_ bundle a private copy of `libstdc++.so.6`  in the AppImage. (`linuxdeployqt` should not have done that anyway.)
* If this is not possible, then maybe using https://github.com/darealshinji/AppImageKit-checkrt/ can help. It will search for `usr/optional/libstdc++/libstdc++.so.6` and `usr/optional/libgcc_s/libgcc_s.so.1` inside the AppImage or AppDir. If found it will compare their internal versions with the ones found on the system and prepend their paths to `LD_LIBRARY_PATH` _where necessary_.

lordofhyphens, 2020-03-20:
@probonopd Part of the problem is that I don't have much in the way of options with Travis. 

I *need* a newer version of gcc than what was on the base install.

bubnikv, 2020-03-20:
> I *need* a newer version of gcc than what was on the base install.

From a poor despair, we are building PrusaSlicer on one docker image with
CentOS 6, because this system still runs on old glibc, while through CentOS
devtoolsets we were able to access a C++17 capable compiler tool chain.

However, we were not able to get AppImage building process running on that
docker with CentOS 6. I suppose that AppImage requires a Debian based
distro to run. So we whipped up another docker, this time with Debian
Jessie and we execute the AppImage building process from the CentOS 6
docker image on the Debian Jessie docker image using ssh.

Does it sound convoluted? Sure. Does it work? Yes. We did not find any
better way to support LTS Linuxes while utilizing C++17.
It was quite time consuming to come up with this solution. We thank a lot
to RetHat / CentOS for the devtoolkit concept, otherwise we would still be
stuck with C++11.

probonopd, 2020-03-21:
>  I suppose that AppImage requires a Debian based distro to run.

No, AppImages should not require a Debian based distro to run. At least the AppImage system itself doesn't - what an application author puts inside an AppImage (and how it is built) is of course up to the application author. Generally we recommend building and testing on the oldest still-supported Ubuntu LTS release because from our experience this satisfies most users; however CentOS 6 with devtoolkit can be a great choice in case you need compatibility with even older systems. We are also using this combination to build AppImageKit itself.

bubnikv, 2020-03-21:
> No, AppImages should not require a Debian based distro to run.

We were not able to get the appimage building process running on CentOS6, but unfortunately I don't have time now to re-run and give you more data.

alfsch1, 2020-03-22:
**Slic3r - AppImage not working at all ! #4946**
Hello,
i tried: 

> DaniDiaz71 commented on 24 Aug 2019
Solved using:

LD_PRELOAD='/usr/lib/x86_64-linux-gnu/libstdc++.so.6' ./Slic3r-1.3.0-x86_64.AppImage

so: it is starting and seem to work ok.

now..i really not understand much of these complex programming/generating an image...
but as far as i can see: just a lib (or 2 ) is missing in the complex of hundrets of files...
so why dont you just include these old c++.so.6 and then it runs on all systems ???
or what is the real problem ??

lordofhyphens, 2020-03-23:
@alfsch1 we did originally, and then it broke newer systems.

alfsch1, 2020-03-23:
Ooo , that should never happen, because the basic intention, to make something like an "app...install" is NOT to have a difficult installation or even need recompile from source (with all the problems), to get a program running on system xy.
just for information: i tried (first) to install slic3r on the "standard" way: (system : MX18.3 , Linux 4.19.5 - amd64 ) debian package with synaptic. Did not work, packet is "defect" . Some days later i tried again, reload new package sources, and install seem working. fine...but: when starting, nothing. no GUI . but command in shell seem ok. so i tried: > slic3r --gui , to see the error.
`$ slic3r   --gui`
`
Can't locate Wx.pm in @INC (you may need to install the Wx module) (@INC contains: /usr/bin/lib /usr/lib/slic3r/ /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/x86_64-linux-gnu/perl5/5.24 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /usr/lib/slic3r/Slic3r/GUI/AboutDialog.pm line 6.
`
but i dont know, what lib is missing , or which Wx module. i was guessing: 
>libwx-glcanvas-perl . but cannot, because "defect" and "conflict" .
so the basic problem seem to be : something wrong in the perl or Wx libs.
what could i try to do ?

lordofhyphens, 2020-03-23:
@alfsch1 anything perl specific gets shipped in the appimage

probonopd, 2020-03-26:
@alfsch1 more on the reasons are documented at https://docs.appimage.org/reference/best-practices.html#binaries-compiled-on-old-enough-base-system.

alfsch1, 2020-03-26:
ok, i see. thx for explainig the problem.
i try install slic3r on my labtop, win10/64 , and working there ok. so i use labtop for slic3r.

Tooniis, 2020-03-31:
this started happening to me when I updated to mesa 20.0.2

autotechmusic, 2020-04-25:
I think MESA, among other things, renamed the drivers just enough to fail on launch.
I'm getting "iris" and "swrast" drivers failing on launch.

AT

lordofhyphens, 2020-06-12:
@bubnikv It's not feasible to include C++17 features in a distro running on something as old as CentOS6 (or 7) and use dynamic linking--you trip over the issues you described.

Especially on CentOS/RHEL, which has the segregated devtoolset setup; you basically shouldn't dynamically link libstd++ or glibc if it requires features from beyond what shipped with the system (4.8.5 for RHEL7).