slic3r/Slic3r #4826: crash after reload model

nesergen, 2019-06-16:
**Describe the bug**
GUI crash after reload rotated model from disk.

**To Reproduce**
Steps to reproduce the behavior:
1. load model
2. rotate model for 90 deg about Y axis
3. right click on model, select "Reload from disck"
4. model updated, then GUI crash. It needs to restart program.

**3D Model and Slic3r Configuration Export**
[BathHolder.stl.zip](https://github.com/slic3r/Slic3r/files/3294495/BathHolder.stl.zip)

**Expected behavior**
Reload model withouy crash

**Desktop (please complete the following information):**
 - OS: Linux Mint 18
 - Version 1.3.0
  

**Additional context**
Another wish. When you update the model, all changes to the model position are reset. That is, the model returns to its original state. It would be desirable that the reloaded model kept the added changes .

Oekn5w, 2019-06-18:
Regarding your additional: I'm currently implementing transformation matrices, which will carry the transformation over to the reloaded object. As I am the last person who touched the reload code I will look into this when I have time.

Oekn5w, 2019-06-26:
I was able to recreate the issue both in downloaded compiled version 1.3.0 and built from the 1.3.0 tag from git (Windows). The crash occurs without anything printed to the console.
But: The crash only occured with the file included in the description. Other STLs (both ascii and binary) I had lying around reload fine in 1.3.0.

I specify the version explicitly because 1.3.1dev (built from commit 48d5b1130130373e990dd40cc811b252556edcb4 ; didn't check auto builds) does not crash with the given file.

@nesergen how did you create / where did you get that file from?

The rotation is irrelevant to the crash.

nesergen, 2019-06-26:
> I specify the version explicitly because 1.3.1dev (built from commit  48d5b11 ; didn't check auto builds) does not crash with the given file.

I use the Onshape.com online CAD service.
--
Отправлено из Mail.ru для Android среда,

Oekn5w, 2019-06-26:
TBH I don't want to create an account to check what might be the root cause for this issue, especially since the development version works.

Oekn5w, 2019-06-26:
and for cross referencing about what I said earlier regarding the rotation: #4401

nesergen, 2019-06-26:
Anyway, thank you for your time.
--
Отправлено из Mail.ru для Android среда,

Oekn5w, 2019-07-06:
@lordofhyphens imo this can be closed

lordofhyphens, 2019-07-07:
Can't reproduce?

closed 2019-07-07 by lordofhyphens