slic3r/Slic3r #4828: Seperate X and Y Path Offsets / Size Compensation

DocTrucker, 2019-06-19:
I have used thingiverse supplied tools for scaling and offset correction and found that the spreadsheet recommends that I need different offset values for X and Y axis.

I am not trying to correct for poorly set x and y axis steps per mm. I have checked physical motion of the axis and it is accurate enough to disregard as a potential cause.

Can we please have the ability to set different x and y axis compensation values in place of the single "XY Size Compensation" variable in the "Advanced" page of the "Print Settings". I would suggest that there is a primary cell and secondary with the secondary defaulting to the same value as the primary in order to not give any back compatibility issues.

I am considering using tools like mesh mixer and blender to try and achieve the same goal, but I believe this is better tackled at the 2D slice level, rather than trying to apply a transformation to the 3D mesh.

Thank you for considering this and the continues development of Slic3r. I look forward to seeing what novel features are in the works!

supermerill, 2019-06-20:
This setting was created to take into account things like plastic retraction (from heat).

I'm not sure of what you try to do.

DocTrucker, 2019-06-20:
I am trying to apply a different fixed offset in  the y and x axis to slice
data.

Scaling is normally used to correct for thermal shrinkage of polymer.

supermerill, 2019-06-21:
You're right on the scaling. Reviewing commits, XY size comp. was created to increase the size of the holes and to allow a more fine-tuning of dimensions.

I don't understand what you want to say by "slice data.".
XY size comp. will shrink /grow the object. As slic3r doesn't do anything different for x and y, I don't see how your printer can make thicker/thinner walls depending on their orientation.

DocTrucker, 2019-06-21:
I have used data measured from a calibration part and found that in order
to get the best dimensional accuracy over a wide range of sizes I need to
apply a  scaling factor and non-symetric offset values for x and y axis.

I'd post on the forums directly if I wanted to try and understand an exact
reason for needing this feature, but given the complexity of polymer flow,
cooling, thermal contraction and the small difference required I feel this
is a pointless exercise. It could even be to non uniform rates of cooling
or angle of part cooling fan.

The list of possible causes is huge. I just want to test the effect of the
non symetric correction.

I have many battles to pitch in the development of my processes and
machines. Right now I have more significant issues to tackle than trying to
figure out why X&Y are different.

lordofhyphens, 2019-06-21:
Afaik scaling in only one direction is nontrivial and I have zero interest
in doing such a thing. I would review a PR for it.

DocTrucker, 2019-06-21:
Non symetrical scaling is an easy single matrix opperation.

Non symetrical offsetting was present on laser path planners for DTM's
sinterstations in the late 90sso I strongly suspect it is little more than
a web search worth of work.

lordofhyphens, 2019-06-22:
I still have zero interest in doing it myself.
I am of course willing to act as a guide in Slic3r's codebase for someone else who wants to work on this, but I will not implement this feature.

DocTrucker, 2019-06-22:
Fair enough. I hadn't looked in to it but was hoping there was more than
one active coder.

I'll build up a case for it in more detail and prove the work with my own
basic slice code, and of course research the other slicers.

What is the normal approach to presenting a feature request rather than the
issue that I have raised?

Thanks.

lordofhyphens, 2019-06-22:
This is the normal way of making a feature request. Interested parties volunteer to work on it. Pull requests (with code) will get sorted out faster.

There are more people than just me, I am just the primary gate to get it into the codebase.

DocTrucker, 2019-06-22:
In that case leave this issue open and I will follow it up with some first
hand data.

lordofhyphens, 2019-06-22:
@DocTrucker yes. That was the plan all along. For what I think is a niche application you have to either persuade me to work on it (or someone else) or submit code.