slic3r/Slic3r #4830: GCode placeholder layer_num contains wrong infomation if two files are printed

yurecz, 2019-06-22:
**Describe the bug**
The value of the placeholder [layer_num] in the generated GCode for **two files** while using the custom G-Code in the "After Layer Change G-Code" section in the "Printer Settings" incremented two times per layer change. Expected behavior that the placeholder should be incremented only once per layer. _If the generation is done only for a single file with several STL copies this issue does not occur._ 

**To Reproduce**
Steps to reproduce the behavior:
1. Add two STL files on the print bed 
2. Add customer G-Code in the "After Layer Change G-Code" section in the "Printer Settings" containing [layer_num] placeholder
3. Generate G-Code
4. Open .gcode file in notepad check the value of the [layer_num] placeholder, use for reference the [layer_z] placeholder to figure on which layer you are right now.
5. The issue here that within the same layer the [layer_num] will be incremented two times, for each file.

**3D Model and Slic3r Configuration Export**
For example, use the following pseudo-code snippet:
;Layer:[layer_num] ([layer_z] mm)

**Expected behavior**
Expected behavior that the placeholder should be incremented only once per layer.
Or It is a feature???!!!

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.3.0

curieos, 2019-07-11:
@lordofhyphens is this a bug? Seems like a bug.

curieos, 2019-07-11:
I can't repeat this in 1.3.1dev, I'll switch over to 1.3.0

curieos, 2019-07-11:
Nevermind that, I think I see what you mean now. It is present in 1.3.1dev as well.

curieos, 2019-07-11:
Looking through the code, it seems like the code for inserting gcode after layer changes is in the wrong function. It's in the PrintGCode.cpp function `process_layer`, I believe this gets called for every object after a layer on that object is finished. I think this logic belongs in GCode.cpp's `change_layer` function, from what I can tell this gets called once per layer, regardless of how many objects there are.

nwneisen, 2019-08-29:
PR #4856 .

I found the layer.id() value would give the behavior expected for non-sequential prints.  This should give the behavior I previously described for sequential prints as well.  I haven't been able to test with a sequential print due to the two objects being placed to close when using the command line.

MaderDash, 2024-06-30:
> inting each individual object, I would expect the _after layer gcode_ for each l

So is this still an issue, or has this been resolved, and this issue is just not been closed yet?