slic3r/Slic3r #4831: Command Prompt Not Working

vatansbeytr, 2019-06-26:
**Describe the bug**
Command console can't open when I try to run  at the command prompt "slic3r-console.exe"
 but program opening not console.

lordofhyphens, 2019-06-26:
Can't reproduce. What are you trying to do?

[image]

lordofhyphens, 2019-06-26:
`slic3r-console.exe` just keeps the background console open when you run it, which it just did for me. 

If you invoke it w/o a path to a STL it will pop up the GUI, as intended.

vatansbeytr, 2019-06-26:
I want to work with console, but I can't slice when I write slic3r-console.exe program pop up and I can't write anything to console

lordofhyphens, 2019-06-26:
It's not a console you type in, it's just there to show you info output from the slicer.

lordofhyphens, 2019-06-26:
Use `slic3r-console.exe --help` to get command line options.

closed 2019-06-26 by lordofhyphens