slic3r/Slic3r #4832: Command Line usage help

vatansbeytr, 2019-06-26:
https://manual.slic3r.org/advanced/command-line
In the link Slic3r can be used as a command line tool. It provides you with great flexibility so that you can perform operations in batch or as part of more complex workflows.
but you said @lordofhyphens I can't type in so which one is true statement.
The real question is this "Is there a way I can load config from settings ini and slice stl file from console ?"

vatansbeytr, 2019-06-26:
I mean how can I use this "slic3r [ ACTION ] [ OPTIONS ] [ model1.stl model2.stl ... ]" in command prompt when I type it doesn't work.

lordofhyphens, 2019-06-26:
Open a cmd.com window.

vatansbeytr, 2019-06-26:
https://i.imgyukle.com/2019/06/26/kGVB5y.png
I know it. I am not an idiot, I am just asking what is the command that I want to load already prepared config.ini and slice it.
Basically I am asking what is the code that make this -->console-->config.ini-->slice-->gcode

vatansbeytr, 2019-06-26:
and if it possible without gui.

vatansbeytr, 2019-06-26:
slic3r C:\Users\xD_xD\Desktop\Slic3r\config.ini C:\a.stl this is just giving error can't load config and not slicing

lordofhyphens, 2019-06-26:
`--load` is the switch to specify config files.

Use `slic3r --help` to see all of the command line options.

lordofhyphens, 2019-07-07:
@vatansbeytr going to close now.

lordofhyphens, 2019-07-07:
````
slic3r --load /path/to/config/file.ini -o /path/to/output.gcode --no-gui /path/to/file.stl
````
on the command line should do what you want.

closed 2019-07-07 by lordofhyphens