slic3r/Slic3r #4833: Load parts from command line

gfournetfayas, 2019-06-26:
Hello !!
i'm working on a project to batch import STL files on Slic3r.
It would be easy, but the thing is that, each object i have to print, contains 4 stl files : 
- The first one is the base of the object
- The three others are linked to the first one and are splitted to be able to assign, to each part, a dedicated density.
From the GUI, i can use the load part menu, after selecting the main part, and i can easily assign to each part, the needed parameters.
I'ld like to be able to do exactly the same thing from CLI, is it possible ?
Thanks a lot for any help !

lordofhyphens, 2019-06-26:
Currently the load part functionality is specific to the GUI.
If someone can figure out a good way to represent it on the CLI I'd be happy to take a PR.

gfournetfayas, 2019-06-26:
OK, thanks for your reply.
I understand that it can be hard to figure it out on the CLI, because additionnal parameters might be added to the parts, same as the main part.
I think about it and come back as soon as possible.
Thanks one more time !

lordofhyphens, 2019-06-26:
If you can figure out a way to do it on the CLI, by all means write it up; it will help a lot in getting something like that implemented.

gfournetfayas, 2019-06-27:
After washing my brain last night, i think that the easiest way is to provide an xml file to the CLI :
-> Call it with (Under win) Slic3r.exe --loadParts file.xml
Here is an xml example that reflects exactly my needs !! 
<parts>
	<main-part>
		<file></file>
	</main-part>
	<part>
		<file></file>
		<bottom-solid-layers>0</bottom-solid-layers>
		<perimeters>0</perimeters>
		<fill-density>10%</fill-density>
	</part>
	<part>
		<file></file>
		<bottom-solid-layers>0</bottom-solid-layers>
		<perimeters>0</perimeters>
		<fill-density>20%</fill-density>
	</part>
</parts>
Also, i'm a profesionnal developper (sh, C, C++, Java, Perl, Python) with 10 years of experience, if i can help i'd be glad to do it :-)
Waiting for your feedback ! Have a nice day.

lordofhyphens, 2019-06-27:
@gfournetfayas We'd also need the relative positioning as parameters.

Slic3r is definitely more of a do-ocracy. I will happily assist in navigating the existing codebase and reviewing PRs, so if you want to work on this go for it.

lordofhyphens, 2019-06-27:
Added tag to attract outside attention.

supermerill, 2019-06-27:
I'm against xml, i prefer json as it's easier to read & parse.

dwillmore, 2019-06-27:
I like the idea of having a plate description language, but I don't have an opinion on xml vs json--I'm not good at either.

I can see myself using this for some of the stuff I do.  Right now I manually arrange a set of parts and save the platter as stl, but that lacks the ability to set different settings for each part, so this would be an imporvement. If @gfournetfayas writes this us and proposes a PR, I would support it.

gfournetfayas, 2019-06-27:
> I'm against xml, i prefer json as it's easier to write & parse.

I don't pay attention to this because i'll use a third part library to generate it, but i can understand that, if created manually, it's important to have a comprehensive language.

@supermerill : Where are you from (for my curiosity !!) ;-) ?

> I like the idea of having a plate description language, but I don't have an opinion on xml vs json--I'm not good at either.
>
> I can see myself using this for some of the stuff I do. Right now I [...]

I'm sorry, but i'm a bit new for collborative work ! What do i do now ?

> @gfournetfayas We'd also need the relative positioning as parameters.
> 
> Slic3r is definitely more of a do-ocracy. I will happily assist in navigating the existing codebase and reviewing PRs, so if you want to work on this go for it.

@lordofhyphens : Thanks for your help and proposal. If i don't make a mistake, the relative position is automatic, not asked when doing that from the GUI; maybe a default value ?

lordofhyphens, 2019-06-27:
What's next is to get a checkout of the source. Fork the repository and make a new branch. Do your work in that branch and then make a pull request against Slic3r/master when it is ready for review.

Keep discussing here about the shape it should take.

lordofhyphens, 2019-06-27:
It is fine to specify relative position compared to the main part in whatever you use.

I believe we have a XML parser already in our dependencies. We do not have a json parser and I would rather not invent one (nor should we have to).

hyperair, 2019-06-28:
Slic3r already has a config bundle format. Can't we just use the same format? You could have one section per file, and the keys in there would specify the overrides necessary for that particular file. Example:

```ini
[file:foo.stl]
bottom_solid_layers = 0
perimeters = 0
fill_density = 10%

[file:bar.stl]
bottom_solid_layers = 0
perimeters = 0
fill_density = 20%
```

bubnikv, 2019-06-28:
The AMF / 3MF formats already support importing all the data required.
Shouldn't one not build upon these formats?

gfournetfayas, 2019-06-28:
> The AMF / 3MF formats already support importing all the data required. Shouldn't one not build upon these formats? pá 28. 6. 2019 v 5:43 odesílatel Chow Loong Jin <notifications@github.com> napsal:
> […](#)

Hello, 
you're right, but sometimes, you don't manage what you receive and it's my case.
Dealing with STL is mandatory for me as i don't have the hand on the previous process.

supermerill, 2019-07-02:
+1 with bubnikv, Why not creating a little utility that package 3D model, config file and transformation into a .3mf?

But the current parser parse only the geometry (as far as i can read), or the one developed for prusaslicer needs to be ported back into mainline.

@bubnikv : why do you add "; " at every line in the config file?

gfournetfayas, 2019-10-22:
Back on it.
I think this answer is the right one, why rewrite something existing ?!
> Slic3r already has a config bundle format. Can't we just use the same format? You could have one section per file, and the keys in there would specify the overrides necessary for that particular [...]

I will start to write code now.
@lordofhyphens : I have some questions to start on the good way, are you a little bit free to help me ? Thanks in adavnce.

lordofhyphens, 2019-10-22:
Start with tests :)

gfournetfayas, 2019-10-22:
> Start with tests :)

In theory you are right, but i never used TDD method.
Could i send you questions in private ? This could be helpull :-)

lordofhyphens, 2019-10-22:
Pretty simple. 

1. Figure out what you want to do.
2. Write a couple simple tests towards the code doing what you want to do. 
3. Make the code pass those tests.
4. Write a few more tests.
5. Goto 3