slic3r/Slic3r #4835: Slic3r gap settings

notsolowki, 2019-07-03:
I'm really sorry for asking this here but I'm trying to slice my stl file and I can't figure out what's causing this gap on the top ledge where the mouse pointer is. If I slice with cura it don't happen. In not sure what's causing this?  
[image]

supermerill, 2019-07-03:
could you take a picture on the layer view?
also, can you link your settings & model?

lordofhyphens, 2019-07-07:
@notsolowki please post a picture from Slic3r's preview, a copy of a 3D Model you are seeing the issue on, which build it is from (see Help->About Slic3r), and a Ctrl-E export of the settings you used to slice.

You can zip them all up and drop them onto github.  Can't do much of anything else until that happens.

lordofhyphens, 2019-07-21:
Closing as it's not reproducible as it is and it's been 2 weeks. @notsolowki post the requested information and I will happily reopen.

closed 2019-07-21 by lordofhyphens

notsolowki, 2019-07-21:
I'm sorry for such a late reply. I too cannot reproduce the error. It's strange. I think it had something to do with the posts I placed in the corners of my model project box. I have since changed the model and not slic3r dont have the problem any more. 1 thing I messed with that changed but didnt fix the problem was the layer height. We can just leave this closed. Thanks for trying to help.