slic3r/Slic3r #4837: Print filename as tag on part

llama, 2019-07-08:
When printing models with many parts (or iterating frequently on a part), it can be frustrating to remember which part is which. My idea is to print a physical "tag" which has the filename on it, to be attached to the skirt or brim of the object. 

This is similar to a feature in Preform which prints the filename on the raft:
https://support.formlabs.com/s/article/Advanced-Support-Structure-Settings?language=en_US#raft-label

I've considered doing this at the CAD phase, but I think it would be best to do at the slicing phase.

lordofhyphens, 2019-07-08:
Would need to make something to convert text to ExtrusionPaths to try to make something like this work. I would consider a pull request.

LawrencePWade, 2019-09-29:
This feature could also be very useful for print farms and educational settings. An old-school dot-matrix printer loaded with sticky labels would be great to print a tag for every model - also won't blow a whole sheet of labels with every item the way a laser or inkjet printer would. 

If you were considering implementing text labels in the model, adding a "Send Model Tag to Printer" feature would be a useful intermediate step while working on the ExtrusionPaths, and would remain useful to some users on a permanent basis. 

I have a 1980s dot-matrix printer that I can use to test such a feature. Anyone wanting this feature for production should be able to find Okidata still makes the ML320 series for exactly this sort of task. 

Hey everyone, thanks for Slic3r. :)

supermerill, 2019-10-10:
In my fork, I added a little header that print the name of each part with their center and their size, in hope that it will be useful for a software that want to know it (like cancelobject or a farm software).

The raft/brim idea is nice but needs a fair amount of effort.
text -> some ascii letter -> conversion-table to polygons -> translation & rotation (HARD part) -> add to brimmed polygons.