slic3r/Slic3r #4838: Unable to upload to OctoPrint from 1.3.1-dev

rmpel, 2019-07-14:
**Describe the bug**
I'm having an issue uploading to OctoPrint in slic3r-1.3.1-dev-8adf792-osx
No error messages, but no uploads either. 
Release 1.3.0 works perfectly, so I know it's not my OctoPrints.
the problem is **not** present in build 1.3.1-dev-5f68ce4.

1.3.0 says ... "generating infill" "saving to /var/..." "sent to octoprint successfully"
1.3.1 (above version) stops after "saving to", does not even attempt sending to octoprint, no mention in octoprint log of any attempt to upload.

Where can I find the log files of Slic3r so I can dig further??

**To Reproduce**
Steps to reproduce the behavior:
Prerequisite: OctoPrint connection set-up and verified with version 1.3.0 (stable)
1. Load a print
2. Click on 'Send to printer'

**Expected behavior**
status bar should report "Uploaded to OctoPrint successfully"
gcode file should appear in OctoPrint

**Desktop (please complete the following information):**
 - OS: MacOS Mojave, MacOS High Sierra (two Macs, both behave the same)
 - Version [e.g. 1.3.1]
1.3.1-dev-8adf792 -> has the problem
1.3.1-dev-5f68ce4 -> does NOT have the problem
1.3.0 (stable) -> does NOT have the problem

As asked above; where can I find log files to investigate further?

lordofhyphens, 2019-07-21:
AFAIK we don't usually log stuff like that. When building from source you can set -DSLIC3R_DEBUG in xs/Build.PL and you will get a lot of verbose output from the slicer itself, but I don't think this part of the GUI is in XS (meaning you'd want to look in lib/Slic3r/GUI/Plater.pm for the code that is actually doing the sending).

lordofhyphens, 2019-07-21:
https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm#L2398 

Looking at `git blame` it looks like that area of the code hasn't changed in years.