slic3r/Slic3r #4841: Slic3r crashes after configuration

ilovelinux, 2019-07-20:
### Version
_1.3.0_

### Operating system type + version
_Archlinux + slic3r-bin (AUR)_

### Behavior
I ran Slic3r and I successfully ended the configuration. I loaded a STL file then I tried to generate the G-Code but Slic3r crashed. Now when I try to start Slic3r I get this error:
```
$ slic3r 
free(): invalid pointer
Aborted (core dumped)
```
If I remove `.Slic3r` folder, it starts up again but re-crash as soon as I end configuration
[.Slic3r folder](https://github.com/slic3r/Slic3r/files/3413908/config.zip)

lordofhyphens, 2019-07-20:
Can you try to use the appimage instead? I don

ilovelinux, 2019-07-20:
I installed slic3r (AUR) which build it from sources. Maybe issue is in binaries downloaded from slic3r-bin?

lordofhyphens, 2019-07-21:
I don't control the slic3r-bin package at all, so I can't help there. I also can't reproduce on my system with https://dl.slic3r.org/dev/linux/slic3r-1.3.1-dev-8adf792-linux-x64.tar.bz2

lordofhyphens, 2019-07-21:
@ilovelinux if the link I mentioned doesn't crash either, I have no idea what that package was doing, and I don't know who the maintainer is for that AUR. You should contact the maintainer of the AUR and direct them here.

lordofhyphens, 2019-07-21:
@ilovelinux Going to close. Not our package and I have no idea how to reproduce (and someone on IRC can't reproduce either). 

If this becomes reproducible with our builds, then can reopen.

closed 2019-07-21 by lordofhyphens

TheAlgorythm, 2020-11-17:
The AUR package just downloads the GitHub Release.

I have the same error.

```sh
$ slic3r
free(): invalid pointer
[1]    54455 abort (core dumped)  slic3r
```

The first time i opened slic3r all was ok until i tried to slice a amf-file. Then slic3r crashed and since then it is not possible to start slic3r anymore.
The only thing which won't crash is slicing from the CLI.
I'm using the AUR binary, which downloads it directly from the GitHub releases page. I'm currently on v1.3.0-3.

Jegeva, 2021-01-28:
Issue here also,

Running on Debian sid directly from the https://dl.slic3r.org/linux/ package (the linux tar)

1)moving ~/.Slic3r to a new location allows slicer to launch again 
2)Strangely the 3 'MySettings.ini' files in the 3 subdirs have the same content.
3)exiting slicer before proceeding with the initial setup doesn't trigger the crash
4)slic3r crashes with the same crash when you try to save a preference file

bref... there is something very wrong with the preference files.

b0ssi, 2021-04-17:
FYI I can reproduce as well, running `slic3r-bin` on Arch. Appimage crashes as well, but as mentioned above, this has most likely to do with the builds in these AUR packages, so to be addressed elsewhere.

TheAlgorythm, 2021-04-17:
@b0ssi 
> But as mentioned above, this has most likely to do with the builds in these AUR packages, so to be addressed elsewhere.

The AUR package downloads the binary directly from GitHub Releases. And I have no idea how a segfault could be caused by eg. a missing library. But that would be something to be investigated.

b0ssi, 2021-04-17:
Right, I've downloaded the binary and Appimage from GH directly to confirm (getting
```
Unable to locate theme engine in module_path: "murrine",
free(): invalid pointer
Aborted (core dumped)
```
The Appimage quits with saying:
```
Unable to locate theme engine in module_path: "murrine",
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_Slic3rxPdiNy/usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/lib/libLLVM-11.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_Slic3rxPdiNy/usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/lib/libLLVM-11.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_Slic3rxPdiNy/usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/lib/libLLVM-11.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1440 error_code 2 request_code 152 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

Let me know if there's anything I could gather from my runtime environment that could give insight.