slic3r/Slic3r #4842: Can't change Perimeter Overhang speed because Bridges settings is grayed out

zapta, 2019-07-21:
**Describe the bug**
Print Settings | Speed | Bridges is grayed out in some conditions, even though it's actually affects the slicing results. This makes tweaking the speed impossible.

**To Reproduce**
1. Load Calicat model from thingiverse  https://www.thingiverse.com/thing:1545913/files
2. Set top and bottom layer counts to 0.
3. Set shell perimeters to 1
4. Set infill to 0%
5. Slice and review the results using Feature Type color mapping
6. Verify that the face of the cat has some Overhang Perimeter type lines
7. Switch the slicing review to Speed color mapping.
8. Now try to change the speed of the Overhang Perimeter lines. Notice that Print Settings | Speed | Bridges is grayed out and don't accept changes.

Actual behavior:  Under the above conditions, Print Settings | Speed | Bridges is grayed out

Expected behavior: Print Settings | Speed | Bridges should be active and allows the user to change Overhang Perimeter speed since it appears in the slicing results.

lordofhyphens, 2019-07-21:
This is controlled in the GUI at https://github.com/slic3r/Slic3r/blob/master/lib/Slic3r/GUI/PresetEditor.pm#L920

With no infill layers, or top/bottom solid layers, it's very specific pathological examples that would have bridge speed.

lordofhyphens, 2019-07-21:
Looks like the only other place that it affects is overhang perimeters.

zapta, 2019-07-21:
Thanks for looking into it. BTW, if both overhang and perimeter overhang use the same settings, what is the reason for presenting them to the user as two different line types?

zapta, 2019-07-21:
(I didn't intend to close. Just clicked the wrong button)

lordofhyphens, 2019-07-21:
Don't know offhand, as it was like that before I got onto the project.

closed 2019-07-21 by lordofhyphens

zapta, 2019-07-21:
Thanks everybody for the quick resolution.

PR #4844 (2019-07-21): Don't disable bridge_speed in GUI when no infill configured.
Bridge speed is still used when overhang perimeters are present, which means that it cannot be disabled if there is no infill. 
Fixes #4842
commit 66af505d (2019-07-21): Don't disable bridge_speed in GUI when no infill configured.

Bridge speed is still used when overhang perimeters are present, which means that it cannot be disabled if there is no infill. 
Fixes #4842