slic3r/Slic3r #4846: Where repieter-server integration?

userosos, 2019-07-24:
Where repieter-server integration? I can not see repetier-server integration.

lordofhyphens, 2019-07-24:
It doesn't exist. Nobody has decided to spend time as of yet implementing it, and as far as I can tell it's closed source.

I have no interest in doing it myself (as I don't use Repetier at all), nor am I really keen on supporting closed source software, especially one that I have zero use for. I also doubt the Repetier team would have any interest in doing the work as it drives people away from their host.

I will look at a pull request, but if they change their API and the integration breaks, it's up for another PR to fix it.

userosos, 2019-07-24:
@lordofhyphens i found info ~2017 year with the code for repieter-server upload.

lordofhyphens, 2019-07-24:
I will look at a pull request. I don't have a test environment set up and do not intend to set it up.