slic3r/Slic3r #4849: V1.3.0 generate corrupted Printer Settings Custom G-code

gwgill, 2019-07-31:
**Describe the bug**
Custom G-code containing if then else syntax randomly looses characters. Generate file is not consistent from run to run.

Example:
 End G-code line is:
 {if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up

and generated files contain:

G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up
G1 Z{z_offset+min(layer_z+30, max_print_height){endif ; Move print head up
G1 Zz_offset+min(layer_z+30, max_print_height)endif ; Move print head up
if layer_z < max_print_height } G1 Z z_offset+min(layer_z+30, max_print_height) } endif }  ; Move print head up

**To Reproduce**
Add line
 {if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up

to Printer Settings Custom End G-code
Export G-code several times to different filenames
Compare the files.

**Expected behavior**
Custom G-code should be generated consistently and copied faithfully.
V1.2.9 doesn't have this problem.

 - OS: MSWindows
 - Version 1.3.0

bubnikv, 2019-07-31:
You are using a G-code macro language of PrusaSlicer, which is different
from the upstream slic3r.

gwgill, 2019-07-31:
OK  - It's not my config., so I was unaware that it was PrusaSlicer specific.
Odd that it gets corrupted randomly though in V1.3.0, and not in V1.2.9.

closed 2019-07-31 by gwgill

lordofhyphens, 2019-07-31:
There isn't a macro processor in 1.2.9.

gwgill, 2019-07-31:
Right, but ideally shouldn't 1.3.0 on parsing incorrect macro's:
1) Emit a warning
2) Produce repeatable rather than random gcode ?