slic3r/Slic3r #4854: The printer has stopped printing (it is working). After the "Print" command, instead of printing, the message: File added to print quere.

Jedrusia, 2019-08-15:
The printer has stopped printing (it is working). After the "Print" command, instead of printing, the message: File added to print quere.
The printer had previously printed without any problems, now I don't want to print any. The printer prints from other programs.
I have version 1.3.1-dev installed together with Repetier. From the Repetier program, the printer prints, but if I run Slic3r without the Repetier program, nothing prints anymore. Earlier this problem was not. I don't want to print through Repetier because it takes more time.

lordofhyphens, 2019-08-16:
Needs more information. What's the printer firmware? How are you trying to use it? 

Need a config file too

Jedrusia, 2019-08-17:
I am using an Ender 3 (Pro) printer connected to the computer via COM 7. The firmware version of the printer is 1.0. Slic3r version 1.3.1 is used in Repetier and everything works here - because the print is controlled by Repetier and Slic3r is only a slice.
If I'm not mistaken (I have not used slic3r for a long time - directly without using other programs) it was possible to print directly from the program on my printer. Now, after loading the model and cutting it (PRINT option), the file lands on the hard disk instead of printing (directly). The printer is set up correctly - I checked it - it is communication with the printer. RepMap (Marlin / Sprinter) is also selected in the printer options.
It seems to me that earlier the prints went directly to the printer and it worked.

Whats congig files you need ?

Jedrusia, 2019-08-22:
Can nobody help me?
Problem diagnosis:
You can't print anything even through the Spoiler.
The printer goes through g-code start codes and that's it. I don't print any model!
However, there are no problems with the operation and printing of models from other programs.

Jedrusia, 2019-08-22:
This is a solution, but author write that only Slic3r have a problem with printing !

Jedrusia, 2019-08-22:
This is a solution: https://github.com/slic3r/Slic3r/issues/4847, but i can't test it now,
Only Slic3r have a problem with printing, any other programs print without problems

lordofhyphens, 2019-08-23:
Patience, please. I am a person working on this stuff in their spare time and I don't use the controller functionality myself.

Jedrusia, 2019-09-10:
Of course lordofhyphens.
I waiting for happy results eliminating any problems.